wled/WLED #3067: E1.31-2018 compliance issues

RichardTea, 2023-01-30:
### What happened?

The current E1.31 reception doesn't quite meet the minimum requirements of the ANSI E1.31-2018 standard.

The issues appear to be:

1. Data using Alternate Start Codes is treated as level data. WLED should only act on Null-Start-Code data, other start codes have other purposes. Some standardised, many manufacturer-specific. 
2. Data marked PREVIEW is acted upon. WLED should ignore any packets marked with the PREVIEW flag. Preview is intended to feed visualisation, and may be sent by one or more controllers at the same time as 'real' level data.
3. Priority is ignored
4. CID is not checked, if there are multiple active sources it flickers rapidly between them all, causing denial-of-service
5. Sources Exceeded behaviour is not stated.

### To Reproduce Bug

Enable E1.31 reception and send data using a test tool such as [sACNView](https://github.com/docsteer/sacnview)

### Expected Behavior

1. WLED should only act on Null-Start-Code data, other start codes have other purposes. Some standardised, many manufacturer-specific. 
2. WLED should ignore any packets marked with the PREVIEW flag. Preview is intended to feed visualisation, and may be sent by one or more controllers at the same time as 'real' level data.
3. Universe Priority to be used to select the winning CID for each universe
4. Data from any CIDs other than the chosen one to be ignored.
5. Sources Exceeded behaviour to be documented as "There Can Only Be One", as this is a very low power device. HTP merging would need far too much RAM, so almost certainly isn't appropriate here.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1 (build 2203150)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

The relevant standards are published here:
https://tsp.esta.org/tsp/documents/published_docs.php

I've started putting together a PR to fix these issues. I don't expect any of them will require configuration changes as any compliant senders will work just fine, and the CID and Priority checks will only make any difference if there are multiple active sources - and as that situation currently doesn't work, it can't make anything any worse!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2023-01-30:
Thank you for the very concise bug report! Most of the points in order to be compliant with the standard seem to be able to be added quite easily.

RichardTea, 2023-01-30:
1 and 2 are trivial and I've already fixed them in a test build, though I've not tested as I'm not yet totally sure how to reflash my ESP32.
I'll probably pop onto the Discord tomorrow for help on flashing my board :)

closed 2024-08-24 by github-actions[bot]