wled/WLED #3070: Railway does not work in 0.14

walfy2007, 2023-01-31:
### What happened?

I think 0.14 broke the railway effect. It looks very different then it used to. 

### To Reproduce Bug

run railway effect on 
0.14 or newer.

### Expected Behavior

same effect as 0.13.3

### Install Method

Binary from WLED.me

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-31:
Can you also provide a video showing the difference between 0.13 and 0.14?

blazoncek, 2023-02-18:
To get alternating colors 1 & 2 select palette "Colors 1 & 2" otherwise it mixes palette colors.

closed 2023-02-18 by blazoncek