wled/WLED #3071: Add possibilty of using fonts with differents widths in scrolling text

ElToberino, 2023-01-31:
**Is your feature request related to a problem? Please describe.**
Small fonts (like "i", j" etc.) cause visible gaps in scrolling text because of the fixed font width.

**Describe the solution you'd like**
Add possibilty to use fonts of variable font widths.

**Additional context**

Using the [MD_MAX72XX library](https://github.com/MajicDesigns/MD_MAX72XX) and the [MD_Parola library](https://github.com/MajicDesigns/MD_Parola) for other LED projects, I like the possibilty of easily adding or editing fonts very much.

Playing around with the new scrolling text fx I successfully tried to add a UFT8-Extended ASCII conversation in order to show characters like ä,ö,ü (Umlaute). Of course I had to edit the console-font file before by adding these characters (their hex values) at the proper position (according to ASCII table).
I got it running - but playing around with the fonts I noticed the visible gaps when narrow characters like "i" or "j" are shown at the matrix. This is caused by the fixed character width.

The libraries linked above can manage a variable font width which makes the text a lot smoother. Another advantage is the possibilty of easily creating or editing complete font sets with online tools like [this](https://pjrp.github.io/MDParolaFontEditor).

I don't know if this would be very complex to implement (I was happy enough to get the UTF8 conversation running) but it would push the scrolling text feature to a higher level. (It also would be nice to show a longer text - perhaps this could be done via a new usermod.)

blazoncek, 2023-01-31:
Only if you are willing to program it.
Fixed width fonts are easy on CPU/MCU and easy to manipulate (less RAM).
Also current fonts (4 at the time of writing) consume a significant amount of flash, adding more would put additional pressure.
IMO not likely to happen.

blazoncek, 2023-02-01:
Related to #3020

keukenrol, 2023-03-31:
If you want more scrolling text functionality, either stream to the matrix using jinx or xlights, or look at the ledtext and ledsprites library: https://github.com/AaronLiddiment

metawops, 2025-01-02:
> If you want more scrolling text functionality, either stream to the matrix using jinx or xlights, or look at the ledtext and ledsprites library: https://github.com/AaronLiddiment You can even give every character a different color using the latter.

Wow! That's nice to see what other repos can do to LED matrices. 
However, I second the OP and would, too, love to see a bit more functionality regarding text in WLED.
Being able to define at least ONE additional, user generated font would be nice. Today's MCUs come with 16 MB of flash memory ...
The Awtrix Firmware that's running on my Ulanzi 32x16 pixel displays has a very nice and perfectly readable 3x5 font. IMHO it's even a bit better than WLED's smallest 3x5 font.
It has even a few Emoji-Unicode-character renderings.

Also, it would be nice to alter the maximum scroll speed a bit more in the "Scrolling Text" effect.
And being able to display the time with seconds would be great – haven't found out, yet, if that's already possible.
Maybe even optionally blinking colons between the HH and MM (and SS) …
And, of course, more than 32 characters would be great, too.

Well, just my $0.02, though.

WLED is *awesome*!! 🙏❤️

ElToberino, 2025-03-23:
I'm still looking for a smooth scrolling text solution for a matrix of 8px height and so I wonder if it would be possible to combine "the best of two worlds".
Meanwhile I took some effort to get behind the "secrets" of The [MD_Maxx72XX library](https://github.com/MajicDesigns/MD_MAX72XX/tree/main), which is the base for the "scrolling text" feature mentioned above. 
The main advantages of this library are the flexible width of the characters and the easily editable font tables enabling (with an additional conversion) the support of UTF-8. This is an important feature for many languages (ö,ä,ü.è,é ....). Of course, there's an increasing need for flash memory, but on the other hand, you can save flash due to the flexible width.
The MD_Maxx72XX library uses font tables with an additional bit determining the character width and offers [functions](https://github.com/MajicDesigns/MD_MAX72XX/blob/main/src/MD_MAX72xx_font.cpp) to handle this value to pick up the proper character out of this table.

My actual plan was trying to mix this up with wled (potentially as a user mod) for my use case ("Simple" scrolling text for some chained 8x8 panels without Y offset, rotation, reverse etc.). I tried to get behind the functions but have to admit that I don't have the necessary knowledge (yet) to implement this, because there is a fundamental difference: WLED draws the characters linewise, whereas MD_Maxx72XX does it columwise (due to the flexible width.)

To make a long story short - I'm not yet able to implement this, but if someone with a deeper knowledge of the wled code has the same need - this could be a good approach. Other things like a special (larger) text array editable via usermod page and/or Json-API could follow and should be easier to implement.

DedeHai, 2025-03-25:
this would need to be a usermod, the way fonts are done in the mentioned library is very "wasteful" on flash usage, a single font file is about 10k. For comparison: a single WLED effect uses about 1k of flash.

ElToberino, 2025-03-25:
Yes, I know implementing this in stock WLED would exceed the limits.
A usermod for such kind of an 8px height matrix setup would also require additional changes like respecting timebase 0 ( = always start the effect from the beginning when it is called), adding a playlist feature that triggers the next effect when the scrolling text was shown completely (1 or x times) and adding a text array other than segment name (more space).

The main advantage of the cited MD_MAX72xx and MD_Parola is (besides the smoother type face) the flexible font file system which allows characters of extended Ascii and enables a simple editing of the fonts with already existing (online) editors. That's why I considered this to be an approach.

I don't know if all these changes can be done in an usermod.... And even if so I'm not capable to do this yet. I'm not a trained programmer and I always try a kind of "reverse engineering" to understand a program and to modify it. This approach often works for me, but the WLED code base is huge...

DedeHai, 2026-04-20:
solved in 16.0 using variable width glyphs.