wled/WLED #3072: Add ESP32-HUB75-MatrixPanel-DMA bus fow WLED

yangminglong, 2023-02-01:
Thank you for providing excellent.  WLED provides an excellent software framework that allows users to customize functions。

**Describe the solution you'd like**
HUB75 is a common LED matrix panel interface,  [ESP32-HUB75-MatrixPanel-DMA](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA) is a popular driver library. It seems possible to add **ESP32-HUB75-MatrixPanel-DMA** as WLED bus to make WLED support HUB75-MatrixPanel。

dosipod, 2023-02-03:
We have used the mrfaptastic DMA DIY project you linked with esp32 only without wled and the wiring picture will only tell you that doing that with wled  only without extra hardware is a long shot . In our DIY project the wifi did not work and the number of pins used is not ideal 
[image]

Might a practical solution is to use something like https://github.com/Pixelvation/Pixelvation with wled

yangminglong, 2023-02-06:
Thank you for your reply. 
To simplify wiring, extra hardware is good, but not necessary.  Wifi should not be a problem.  I have followed [mariobros-clock](https://github.com/jnthas/mariobros-clock) and [RGB-Matrix-LED-Mario-Bros.-Clock](https://github.com/James-workshop/RGB-Matrix-LED-Mario-Bros.-Clock) project. WiFi works normally. NTP time can be obtained through WiFi.

Wiring, Recommended by ESP32-HUB75-MatrixPanel-DMA
```
Wiring the ESP32/ESP32-S2/ESP32-S3 to an LED Matrix Panel

If you want to change the GPIO mapping at runtime, simply provide the wanted pin mapping as part of the class initialization structure. For example, in your sketch have something like the following:

// Change these to whatever suits
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 19
#define C_PIN 5
#define D_PIN 17
#define E_PIN -1 // required for 1/32 scan panels, like 64x64px. Any available pin would do, i.e. IO32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
HUB75_I2S_CFG mxconfig(
	64, // Module width
	32, // Module height
	2, // chain length
	_pins, // pin mapping
);
dma_display = new MatrixPanel_I2S_DMA(mxconfig);

```
[Schematic_LED on url](https://github.com/James-workshop/RGB-Matrix-LED-Mario-Bros.-Clock)
[image]
[image]

dosipod, 2023-02-06:
Interesting you can use wifi ,we only tested mrfaptastic  DMA project a while back  and did faced few issues with wifi but could be old build / info or we had an issue in our code . We will try to test multiple projects including  the one  you linked . If wifi is usable  then could it be  possible to  add ArtNet or DDP support to your project ? if so then we can use wled on another mcu  or even use something like xlights for animation and effects . I hope the ones you linked are easy to replicate

dosipod, 2023-02-06:
Seems to me the pics you listed are from other project and not yours , if you personally tested wifi and it worked then show  the code. 
[image]

witnessmenow, 2023-03-18:
@dosipod I've done a lot of work with the esp32 and Hub75, mainly just using the mtfapptsstics library. 

I'm happy to run some tests if it would help, just let me know what you want me to try and I'll let you know 

I have designed some open source hardware to connect to that makes that part a lot simpler (esp32 Trinity ), I'm also happy to provide one to a maintainer of wled if it would help the process

dosipod, 2023-03-20:
Setup mtfapptsstics  project and access wifi at same time . create a web server and see if it is stable  . One your test that then share pics and video of the same along with code used

witnessmenow, 2023-03-20:
Ok sure.

I have definitely made projects in the past with a webserver and the
matrix, but I'll make a minimum viable project and share it here.

witnessmenow, 2023-03-20:
Here is a video of a project that has a web server running on an ESP32 while also controlling the matrix.

https://youtu.be/qho0EJC-cqE

The project is a modified "HelloServer" from the ESP32 webserver examples, combined with an example I have for just scrolling text.

Code for the example is here: 

https://gist.github.com/witnessmenow/b12decb79b4c9c782d1c0af4c43559b6

Is this what you were looking for @dosipod ?

dosipod, 2023-03-21:
Which hw you have use for that ?"ESP32 Trinity ?
Did you test on a regular esp32

witnessmenow, 2023-03-21:
Yeah that's using a Trinity. 

I'm not sure what you mean by "regular esp32"?

The Trinity is just a uniquely shaped esp32 dev board, it's nothing special. It's just a wroom module with an appropriate connector broken out for connecting to the display. All it has other than that from an esp32 point of view is a USB to serial and a 3.3v reg. Doesn't have any psram or anything like that.

Anything you can make with a Trinity you can make with any esp32 with enough pins broken out. Here is a wiring diagram that is equivalent logically to the Trinity.

[image]

As you mentioned in your earlier comment about doing it diy, there is a lot of connections and it's a bit of a pain, so the Trinity just makes it easier.

Trinity is fully open source so if people want to make their own, they are welcome to.

dosipod, 2023-03-21:
I trust what you are saying as you have showing in the video , only small matter that Trinity board cost $30 but  a cheaper option that work with wled out of the box is https://github.com/Pixelvation/Pixelvation . During our limited testing  which could have  shortcoming of  the   wiring as you see in the pic we posted  it was not stable with wifi . We have tested that almost one year ago and might try to do that again with the updated code  but honestly not looking forward to wiring mess again

witnessmenow, 2023-03-21:
Looking at the Pixelvation, it and the Trinity are fundamentally the same, so I may try get pixelvation engine running on the Trinity as an intermediate solution.

Any progress made on this would not be Trinity specific, it could even be used with pixelvation directly, rather than it emulating APA102 from a second ESP32, or with any ESP32 if your patient enough to wire it up 😄 

As mentioned earlier, I'm happy to provide a trinity to a WLED contributer for working on this, with no expectation of results or even progress. (any work at all even!)

From a quick look, without fundamentally changing how WLED works, adding support the matrix panels would make more sense at the NeoPixelBrightness level? If one was to create a new LED type for NeopixelBrightness, which interfaced with the HUB75 matrix, then getting it working in WLED would be straight forward. 

I only took a quick look at everything at lunch time so I could be mistaken. Is there LED abstraction layer I missed in WLED?

dosipod, 2023-03-22:
Some the guys are asking if you could show the video running mtfapptsstics  effects along side the web server as you only showed text and not effects . Also the code you linked to is not complete so the guys could test it . attache the whole project  . And we do have our own HW but mainly work with  normal MCUs and noting that is costly so that others could easily replicate

witnessmenow, 2023-03-22:
That code is the complete project? There are two dependant libraries, mtfapptsstics matrix library and adafruit gfx. Other than that, just fill in your ssid and password. 

The only other thing I can think of is the pins for matrix, I use the default pins of the library so I don't need to set them. If you are using different pins you can set them in display config like I have for the e pin on like 72.

I can add a webserver to the effects demo if you want, not sure what it will prove? The text demo is already a constantly updating project already as it needs to scroll the text

Ok, I get it, you think the Trinity is too expensive 😅 let's remove the Trinity from the discussion

witnessmenow, 2023-03-22:
Added HelloServer code to Aurora:

Video: https://www.youtube.com/watch?v=9gHJLBsv8c8

Code: https://github.com/witnessmenow/AuroraPlusWebserver-WLEDTest

Some Instructions:

1. Install 3.06 of MrFaptastic's library (seems to be an [issue with 3.07](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/426))
2. Install fastLED (this sketch depends on it)
3. Open the "AuroraDemo" from the matrix library demo.
4. Get that running on your matrix (you may need to change pins etc) - No point involving Wifi til the sketch works without it.
5. When that works, apply whatever changes you made to my sketch ( Or add my changes to the example, search for "added by Brian"
6. Update the Wifi details

dosipod, 2023-03-24:
Very interesting , few days ago i was in the middle of setting up colorlight card and  hub75 with FPP with the aim to use that with DDP  and wled. . Since the webserver is fine do you think it would be possible to setup the MrFaptastic unit as DDP or artnet receiver .  This way we could use all wled effects . Also this is the code for audio sync might be useful here https://github.com/netmindz/WLED-sync . We have 4 different projects at the moment with hub75 and might change the direction for that  based on your good outcome , We are on an offsite  project at the moment but for sure will test that in more details after. The price of HW is a bit of a concern as we work with students but  we are also testing with PIs and receiving cards so might be Trinity cost is reasonable ( i see $30 vs $5 esp32 is a bit crazy but i might still try  to get it  if we can get other hw with  it)

mzashh, 2023-04-26:
why even bother with using trinity, if you think its too expensive you can diy something like this, like what i did

[image]
[image]

as for WIFI it works fine in my experience, here is some code which uses WiFi and hosts a Web-UI

https://github.com/mzashh/HUB75-Pixel-Art-Display

the only thing to worry about is the panel variation as there are 100's of panel variations and not all are compatible with the library

dosipod, 2023-04-29:
Yeah we have mrfaptastic  project with standard ESP32   as you see in the pics , performance  was not bad but not that great  , I was looking for a project that uses a web server in a stable way  Might since your project is newly updated ( still Asynchronous  )  that you  could  test it as an  artnet or ddp receiver  Still interesting Pixel-Art-Display

yangminglong, 2023-05-24:
https://github.com/FastLED/FastLED/blob/master/examples/SmartMatrix/SmartMatrix.ino

RagnarLoadbrok, 2023-11-16:
What about Matrix Portal from Adafruit? That and the Divoom App would be awesome

mzashh, 2023-11-16:
> What about Matrix Portal from Adafruit? That and the Divoom App would be awesome

The matrix portal isn't much different than using a standard ESP32, its pretty much the same thing except for the convenience of wiring up the matrix.

As for Divoom, their stuff is proprietary and most of their products use obscure mcu's like the ANYKA AK1052 except for the Divoom pixoo max (64x64) which uses an ESP32 in combination with an ARTERY AT32 (stm 32 clone). so basically we cant use their app even if we reverse engineer it, it would be illegal to use their stuff.

tho i am working on a similar pixel art display, and have lots of plans for it, but die to my current circumstances I don't have enough free time to work on it even though i have a lot of features almost done. here is the repo link if you are interested.
https://github.com/mzashh/HUB75-Pixel-Art-Display/

RagnarLoadbrok, 2023-11-16:
> > What about Matrix Portal from Adafruit? That and the Divoom App would be awesome
> 
> The matrix portal isn't much different than using a standard ESP32, its pretty much the same thing except for the convenience of wiring up the matrix.
> 
> As for Divoom, their stuff is proprietary and most of their products use obscure mcu's like the ANYKA AK1052 which have poor documentation except for the Divoom pixoo max (64x64) which uses an ESP32 in combination with an ARTERY AT32 (stm 32 clone). so basically we cant use their app and even if we manage to reverse engineer it, it would be illegal to use their stuff.
> 
> tho i am working on a similar pixel art display, and have lots of plans for it, but due to my current circumstances I don't have enough free time to work on it even though i have a lot of features almost done/planned for it. Here is the repo link if you are interested. https://github.com/mzashh/HUB75-Pixel-Art-Display/

thanks for your answer, ill be waiting, maybe someone clones the software, would be comfortable to use instead of coding everytime to change the screen ;)

mzashh, 2023-11-16:
> thanks for your answer, ill be waiting, maybe someone clones the software, would be comfortable to use instead of coding everytime to change the screen ;)

that's pretty much what my software is, for now you can upload multiple GIF files on it via a web page and play each one or let them loop one by one, no need to reprogram software.

RagnarLoadbrok, 2023-11-16:
> that's pretty much what my software is, for now you can upload multiple GIF files on it via a web page and play each one or let them loop one by one, no need to reprogram software.

Ok  i will test it, thank you

dosipod, 2023-11-16:
I am not really sure of the progress as of now as i did not have a chance to test but someone is working on native hub75 support with wled https://github.com/MoonModules/WLED/pull/87, might try it out

Jfeif27, 2023-12-28:
Hey all. I was curious if anyone got this working with WLED. I'm trying to use a third party software called SignalRGB and if I can get hub75 to be controlled by wled that would be awesome. Thanks for all the previous work on this.

netmindz, 2024-05-26:
Support has been added to MoonModules - https://github.com/MoonModules/WLED/pull/108 and https://github.com/MoonModules/WLED/releases/tag/untagged-85eec230baeafee03968

There is also a MR open to add to the AirCookie version of WLED - https://github.com/Aircoookie/WLED/pull/3777

netmindz, 2024-11-05:
This is coming to AC shortly 

https://github.com/Aircoookie/WLED/pull/3777

closed 2024-11-05 by netmindz

huytran-df, 2025-11-12:
> Yeah that's using a Trinity.
> 
> I'm not sure what you mean by "regular esp32"?
> 
> The Trinity is just a uniquely shaped esp32 dev board, it's nothing special. It's just a wroom module with an appropriate connector broken out for connecting to the display. All it has other than that from an esp32 point of view is a USB to serial and a 3.3v reg. Doesn't have any psram or anything like that.
> 
> Anything you can make with a Trinity you can make with any esp32 with enough pins broken out. Here is a wiring diagram that is equivalent logically to the Trinity.
> 
> [image]
> 
> [link to Full size to actually read the pin numbers](https://github.com/jnthas/clockwise/blob/gh-pages/static/images/display_esp32_wiring_bb.png)
> 
> As you mentioned in your earlier comment about doing it diy, there is a lot of connections and it's a bit of a pain, so the Trinity just makes it easier.
> 
> Trinity is fully open source so if people want to make their own, they are welcome to.

I have consulted a document that guides me how to connect ESP32 to MatrixPanel but I see that the wiring of this website is different from yours at pin B and E. Can you help me with the correct way? Thanks

[image]

DedeHai, 2025-11-12:
please use discord or the discourse forum for support, github is not the place.