wled/WLED #3074: usermod 4LD_ALT + Rotary_ALT: display garbled on esp32

softhack007, 2023-02-02:
### What happened?

When 4LD_alt and rotary_alt usermods are used together on esp32, the display can look "corrupted" when rolling through options using the rotary encoder.

I am working on a fix. The root of the problems is that display calls are made concurrently - 4LD_alt has a dedicated redraw task, while rotary_alt does direct calls of display methods. These can interfere. As U8x8 is not thread-safe and has no interlocks for draw calls, this must be done (using transactions protected by mutex) in the usermod.

A temporary (but still incomplete) fix is in the MM fork, see https://github.com/MoonModules/WLED/commit/e1f14628b03ff21623a149251246b759c823099e

I will bring the fix into upstream AC WLED once its stable.

### To Reproduce Bug

set up 4ld_alt and rotary_alt
use rotary to change settings visible on the OLED
* when overlays appear, the main display starts to be corrupted. Sometimes WLED crashes
* "rolling" through palettes / effects can also leads to display corruption.

### Expected Behavior

display remains readable, no crash

### Install Method

Self-Compiled

### What version of WLED?

WLED main development branch, or WLED MM/mdev fork

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-S3, ESP32-S2, ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-02-02:
I also see panics (in I2C core functions) and reboots.

blazoncek, 2023-02-02:
The easiest solution is removal of task for updating display and relying on usermod loop() to redraw display when needed.
Task was experimental to see if it benefited audio processing.

softhack007, 2023-02-02:
> Task was experimental to see if it benefited audio processing.

Actually I like the task solution, as it does help for audio processing. Maybe i find something to make it work. 

As an intermediate workaround, you're right we could go back to usermod loop() for display redraw.

softhack007, 2023-03-10:
I think I found a fix, basically putting semaphores around display calls.
Need to do some more testing, and then I'll bring it into the main source code.

blazoncek, 2023-03-11:
Yes, I've seen your updates in MM but it adds a lot of complexity IMO and other usermods need an update too to take full advantage.

I would remove `canDraw()` method altogether and rely on internal logic of 4LD to update display at appropriate time (or perhaps cancel current drawing if possible if new request comes in).
Perhaps a better approach (though using more RAM) would be to make a display queue or wait until drawing is complete (waiting may add unwanted consequences).

My original intention (with drawing task) was display queue. So anything that needs to be written to display would only add to the queue (returning immediately) and only display task would draw to the display.

blazoncek, 2023-04-24:
@softhack007 do you still see corrupted display after my update?
If no, please close.

blazoncek, 2023-08-05:
@softhack007 any updates?

softhack007, 2023-08-05:
@blazoncek I still want to add semaphores to the critical draw calls, similar to what was done in MM fork. After that the changes should be "bullet-proof" (see my older DM in discord).

I'd like to keep this bug report open just as a reminder for myself to do it.

blazoncek, 2023-08-17:
@softhack007 a kind reminder. 😄

softhack007, 2023-08-20:
> @softhack007 a kind reminder. 😄

Hi, i've tried hard today, but could not reproduce a crash any more 🥇
So I agree to close this bug report.

I would still like to add a few semaphores to make the thing bullet-proof, but that will be a separate PR coming in the next days.

closed 2023-08-20 by softhack007