wled/WLED #3077: M5Stamp ESP32 PicoD4 boot loop?

swissfreek, 2023-02-04:
### What happened?

Something in the 0.14.0b1 firmware doesn't like the M5Stamp ESP32 Pico. This is the version that doesn't have USB or a screen or anything.

It detects and installs fine using the ESP32 downloader, but after the install finalizes and it reboots, I assume it goes into a boot loop, because it never reconnects to the install page (to do the WiFi settings, etc.) and the WiFi AP never shows up.

I can install 0.13.3 just fine, and it works as expected, to include addressing the built-in WS2812 on the board.

### To Reproduce Bug

Install 0.14.0b1 onto an M5Stamp ESP32 Pico D4 stamp

### Expected Behavior

It installs and then reboots and requests WiFi settings in the installer.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0b1

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-03-11:
Please connect your device to a computer using a USB cable.
Then open a serial monitor (115200 baud), for example in Arduino IDE or in platformIO.
Post the serial monitor output here, so we can have a look.

swissfreek, 2023-03-11:
When installing 0.14.0-b1, I get this after reboot in an endless loop (this is pulled from the WLED console viewer on the install site; when I tried to use a serial monitor with ArduinoIDE I got a lot of junk characters and it wasn't legible):

_Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40142828  PS      : 0x00060130  A0      : 0x80140a5e  A1      : 0x3ffb1da0  
A2      : 0x3ff440c0  A3      : 0xfffffffb  A4      : 0x00060120  A5      : 0x00000001  
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x00060120  A9      : 0x3ffb1d80  
A10     : 0x00000000  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x0000001e  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40142825:0x3ffb1da0 0x40140a5b:0x3ffb1dc0 0x40140de2:0x3ffb1de0 0x400e91b3:0x3ffb1e10 0x400e932a:0x3ffb1e60 0x400ec999:0x3ffb1e90 0x400ecb9e:0x3ffb1eb0 0x400ed1cd:0x3ffb1ef0 0x4010c6a5:0x3ffb1f20 0x4010caae:0x3ffb1f40 0x4010d97a:0x3ffb1f90 0x4012e4a2:0x3ffb1fb0 0x4008b6ae:0x3ffb1fd0

Rebooting..._

**BUT.** If I install 0.13.3, when the install completes it asks for WiFi credentials and the serial monitor says this once:

_ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada

>_

So 0.13.0 seems to install and function properly, but something in 0.14.0-b1 is causing an issue. I see mention of improvements for Pico controllers in the 0.14.0-b2 notes, but waiting for that binary to be released to find out if the issue has already been corrected.

softhack007, 2023-03-17:
There is a chance that your problem is related to issue #3128, as your board might not have a serial-to-USB chip for uploading via USB. If the Serial RX pin is "floating", then WLED will receive random commands and become erratic.

If my idea is right, you might be able to use the M5Stand by first configuring buttons on GPIO 1 and GPIO 3, then updating to WLED 0.14.

swissfreek, 2023-03-17:
The M5Stamp most definitely does not have an onboard USB-to-Serial chip. You have to use an adapter to upload code. I will give the button idea a shot.

swissfreek, 2023-03-18:
That worked. I was able to load 0.13.3 on it, set buttons as you suggested, and then do an OTA update to 0.14.0-b1, and all seems to be working properly. So my issue is either the same as #3128 or at least solved the same way.

blazoncek, 2023-03-18:
Then close, please.

closed 2023-03-18 by swissfreek

gibsonplayer52, 2023-04-10:
can i ask how to "set the buttons as you suggested" is done? im trying to piece this convo together but i dont know what the fix here is referring to. Im having the same issue with my m5stamp pico.  hope his is the correct place to leave this question,m im pretty new to all of this. Thanks in advance

blazoncek, 2023-04-10:
@gibsonplayer52 please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

csaleman, 2023-10-07:
@softhack007 setting the buttons worked for me too. TY