wled/WLED #3078: ESP32-C6 is not supported

Enelar, 2023-02-04:
### What happened?

Unable to flash WLED firmware to ESP32-C6-DevKitC-1 v1.1

### To Reproduce Bug

Just try to flash wled firmware to any ESP32-C6 chip

### Expected Behavior

I guess we need two versions of firmware

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell
[user@user-nvme esp32]$ sudo esptool.py --no-stub write_flash 0x1000 ./esp32_bootloader_v4.bin 
esptool.py v4.4
Found 2 serial ports
Serial port /dev/ttyS4
Connecting......................................
/dev/ttyS4 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (revision v0.0)
Features: WiFi 6, BT 5
Crystal is 40MHz
MAC: 60:55:f9:f6:a2:bc
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00010fff...
Erasing flash...
Took 0.29s to erase flash block
Wrote 65536 bytes at 0x00001000 in 0.6 seconds (900.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[user@user-nvme esp32]$ sudo esptool.py --no-stub write_flash 0x10000 ./WLED_0.14.0-b1_ESP32.bin 
esptool.py v4.4
Found 2 serial ports
Serial port /dev/ttyS4
Connecting......................................
/dev/ttyS4 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (revision v0.0)
Features: WiFi 6, BT 5
Crystal is 40MHz
MAC: 60:55:f9:f6:a2:bc
Enabling default SPI flash mode...
Configuring flash size...
Unexpected chip id in image. Expected 13 but value was 0. Is this image for a different chip model?

A fatal error occurred: ./WLED_0.14.0-b1_ESP32.bin is not an ESP32-C6 image. Use --force to flash anyway
```
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-02-04:
Correct observation.
There are no plans to support C6 at the time as C3, S2 and S3 are still in early experimental phase, not yet ready for production use. And those are older, more mature chips.

softhack007, 2023-02-04:
Similar for other very brandnew chips: [-C2](https://www.espressif.com/en/news/ESP32-C2), [-C5](https://www.espressif.com/en/news/ESP32-C5), [-C6](https://www.espressif.com/en/news/ESP32-C6_Available) or [ESP32-P4](https://www.espressif.com/en/news/ESP32-P4)

To put it into some context - availeability of ESP32-C6 was announced just 3 weeks ago on Jan 6, 2023. Knowing espressif, I bet it will take **at least 2 years** until stable development environments and open source libraries are ready for any "serious use" on -C6.

We shoulf put it on the backburner, until at least stable support is availeable for the core libraries  we need:
* ESP-IDF
* arduino-esp32
* Neopixelbus 

For now, we have enough troubles to make -S3/-S2/-C3 work. So don't expect support for other new models soon.

cherrydev, 2023-03-30:
Supporting those newer chips requires updating to the esp-idf v5.1. From what I've seen, the migration documentation is pretty good. A potentially valuable bonus is that some of the newer chips, like the S3, now support DMA over the RMT interface. From my experiments, enabling DMA reduces the CPU usage during a transmission to 0.01% compared to about 18% using PIO. I believe this would also reduce the chances of a rogue interrupt causing a glitch proportionally. I'm not sure if this project bumps up into those limits or has a problem with interrupt-caused glitches, but if so, that might be another reason to consider it.

blazoncek, 2023-03-30:
A project as big as WLED needs manpower to support various chips, not just development environment.
Any help would be appreciated.

cherrydev, 2023-04-01:
I'm not a user of WLED but also write LED drivers for ESP32, and I'm not making requests, demands or suggestions that you do anything at all. I was sharing information I'd learned about the new features in ESP-IDFv5 that might be relevant to someone else writing LED driver code on ESP chips, since it offers a significant performance improvement. I was excited by my discovery and thought the people working on this project would be too. Your project is impressive and I hope you find the help you're looking for.

blazoncek, 2023-04-01:
Thanks for your willingness to share.
Unfortunately we do not write LED (or any other) drivers ourselves but reuse existing @Makuna 's excellent NeoPixelBus.

As such there is (almost) nothing (apart GPIO numbers related stuff) in WLED core code that is chip specific. Libraries as NeoPixelBus or LittleFS and AsyncWebServer are the ones that would need to add support for the above mentioned chips before WLED can support them as well

Though some usermods may use particular HW but those are optional and not all support every chip.

cherrydev, 2023-04-02:
I've been over to the NeoPixelBus repo and looking at Makuna/NeoPixelBus#598 it seems they're already looking into it.

aaalek, 2024-10-05:
It's been some time since this issue has been created, have there been any advancements with the libraries and such to make the new chips work?

softhack007, 2024-10-05:
> It's been some time since this issue has been created, have there been any advancements with the libraries and such to make the new chips work?

Not enough advances, unfortunately - it looks like my "at least 2 years" estimate was quite accurate.. 

If NeoPixelBus would be able to utilize the new RMT driver in esp-IDF 5.1.x, we could make an attempt to get WLED running on the -C6. Maybe this will happen during the next year. For other new chips - nothing in sight.

Makuna, 2024-10-06:
The new RMT support in the Core3 branch.  Still not configured to use the new boards yet.

softhack007, 2024-10-19:
> Sorry to be that guy, but is there an rough estimate on how long it will take? I'm looking to invest in to these boards, once support is added. Thanks

As you are "that guy" you probably guessed the answer 😉 
"it's ready when its done". 

A bit more seriously - there is no schedule for supporting the -C6 in WLED. Personally I have a fork that compiles for -C6, but there is still a lo g way to go.
C6 support _might_ come as "experimental" in 0.16 (one year or more), and it will be comparable to the -C3 performance wise.

From experience with other "new boards" like the -S3, I would expect that we'll need around two years until WLED runs stable on this MCU. 

So if you're looking at a project or other abusiness case, I'd say don't bet you money on the -C6.

pafftis, 2024-10-22:
I'm a bit confused. You write that the esp32-c3 and esp32-s3 boards, the mini ones, are not supported normally yet. But still, do you have some firmware posted, are these boards supported or not? If yes, do they work well?

DedeHai, 2024-10-22:
C3 and S3 are *mostly* supported now but still not as stable as classic ESP32 and there may still be undiscovered bugs. Not all variations of C3 and S3 boards are supported by the installer or WLED, you may need to edit code or manually install.

amcewen, 2024-10-22:
> A bit more seriously - there is no schedule for supporting the -C6 in WLED. Personally I have a fork that compiles for -C6, but there is still a looooong way to go.
> C6 support might come as "experimental" in 0.16 (may be released in one year), and it will be comparable to the -C3 performance wise.

What would be the best way to help out with -C6 support?  I should have a few -C6 boards to try out in a couple of weeks, and should be able to contribute some development effort.

blazoncek, 2024-10-22:
> You write that the esp32-c3 and esp32-s3 boards, the mini ones, are not supported normally yet

Nobody said that. MCUs, C3 and S3 (and S2) are well supported in WLED. The issue is development board implementation by different hardware manufacturers. There are variants that will not play well with officially distributed versions of WLED binaries. In such case you'll need to compile (and figure out compilation parameters) yourself. There are also poorly designed boards out there. No software can help with that.

> do you have some firmware posted, are these boards supported or not

Yes we have. Yes and no. Depends on particular development board. See above.

> If yes, do they work well?

Depends on what you consider "well". There may be connectivity issues if your WiFi is not perfect, you may see performance lag compared to ESP32, etc. As there are numerous development boards available and not all are made equal we cannot support *all* of them. Currently working "well" are the following (the ones I tested, @softhack007 and @DedeHai may add theirs):
- Lolin ESP32-S2 mini (single core, low on RAM, direct replacement for Wemos D1 mini)
- LilyGo TTGO T-01 C3 (ESP32-C3, single core, slow speed, plenty of RAM, direct replacement for ESP01)
- Espressif ESP32-C3-DevKitM-1 (ESP32-C3, single core, slow speed, plenty of RAM)
- Lolin C3 pico (ESP32-C3, single core, slow speed, plenty of RAM, direct replacement for Wemos D1 mini)
- generic ESP32-S3 Zero (dual core, fast, plenty of RAM, bad WiFi performance, gets hot, bad effect performance)
- generic ESP32-C3 Zero (single core, slow speed, plenty of RAM, poor WiFi performance)

blazoncek, 2024-10-22:
> What would be the best way to help out with -C6 support?

Compile until it works. When compile works, figure out GPIO restrictions, RAM restrictions, library restrictions.

DedeHai, 2024-10-22:
I can add this to the list
- C3 Supermini - There are two variants, one works, one has very (!) poor wifi, there is a reddit post about it

Personally I do not see any benefit in the C6 over C3 for WLED, it just has more GPIOs but so does the S3.

blazoncek, 2024-10-22:
C3 is far from ideal. But it is a good start.
The smallest development board I've seen is Seed Studio XIAO S3.

Depending on what you need, @srg74 may have LilyGo TTGO T-01 C3 compatible board which is 30x20x17 mm including LilyGo board.

softhack007, 2024-10-22:
> I'm looking for the smallest board possible to put a levelshifter on and connect it to the strip. My maximum profile width is 3 cm and the depth is 2 cm. Esp32-c3 is ideal - it just fits, all other boards are bigger unfortunately.

The smallest one I know about is the waveshare [S3 Zero](https://www.waveshare.com/wiki/ESP32-S3-Zero) - 1.8cm x 2.5cm. It needs the [`esp32s3_4M_qspi`](https://github.com/Aircoookie/WLED/blob/6d1126b8aa123ebf351bb73b2618ccfcb9199682/platformio.ini#L558-L576) buildenv.

netmindz, 2024-10-22:
Yeah I wanted to try the C6 for ZigBee, but it's looking like I might end up writing a usermod to communicate with a C6 from a supported ESP32 before having a WLED build running on the C6 alone

sapkra, 2024-10-22:
@netmindz Actually having a similar idea but instead of writing a usermod I will just remote control WLED by simulating a ESP-NOW remote on my C6. I need the C6 because it has 2 TWAI controllers.

pafftis, 2024-10-23:
> The smallest one I know about is the waveshare [S3 Zero](https://www.waveshare.com/wiki/ESP32-S3-Zero) - 1.8cm x 2.5cm. It needs the [...]

Is this particular board currently supported or not?

softhack007, 2024-10-23:
> Is this particular board currently supported or not?

Please define "supported". I have one on my desk and I can upload WLED 0.15 to it - from the developer environment (VSCode+platformIO). I'm not sure if the web installer has all the necessary files. But you may be lucky. Please remember it is a raw ESP32-S3, so you'll need to add level shift and other components for driving your LED strip.

This is an open source community project - you can't get a definite guarantee from us. My suggestion would be to order one, and try it out. 

> Контроллер WLED ESP32C3 5-24 В, https://sl.aliexpress.ru/p?key=EAM5rXK
> Here on Ali they sell some ready-made boards. Are they bad?

I have no idea. It's a C3  so its very minimal, and no audio support. Maybe ask the seller how to upload WLED updates (USB preferred).

> ### we are drifting off-topic.
>  If you have further questions about what kind of board to chose, there is a very friendly WLED community in discord. Please ask for help on discord   [image]

pafftis, 2024-10-25:
Do I understand correctly that the main problem is the lack of libraries during the firmware of the board? If the firmware is loaded somehow into the esp32-c3, then most likely it will work fine?

blazoncek, 2024-10-25:
> If the firmware is loaded somehow into the esp32-c3, then most likely it will work fine?

"Somehow" is by using web installer or PIO upload or manually uploading using esptool. As mentioned several times C3, S2 and S3 are fully supported in firmware. Getting correct binary is somehow tricky for S3.

amcewen, 2024-10-25:
@softhack007 thanks for sharing your experiences with -C6 and take on where things are build-wise.

I hadn't seen the platformio / espressif politics until now - my other big ESP32 project at the moment is using esp-idf directly (with Arduino as a component).  It does seem like -C6 support would only make sense once it's clearer how WLED is going to address things w.r.t. platformio and arduino-esp32 v3.0.x

I'll probably go with a -C3 for now then (the Bluetooth 5.3 support on the -C6 was what was tempting me that way, but was always a longer-term thing-to-play-with :grin:)

As I say, I'm not tracking platformio closely and so might be looking in the wrong place, but it looks like they're still supporting later versions of esp-idf?  Would moving to [Arduino as a component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html) be a path to migrate to newer versions, and arduino-esp32 v3, while continuing to use platformio?

I've got the move from esp-idf v4 and arduino-esp32 v2 to v5.3 and v3 looming on another project, so will have a bit of experience with that move (with arduino as a component) in a month or so.  On a personal note it's a shame to see `WiFiClient`, etc. deprecated, given that I introduced them when I was in charge of the original Arduino Ethernet library :cry:

damnms, 2024-11-05:
The c6 would be very nice for me because of those 2 reasons:
1) here is hell with 2.4ghz networks
2) my router can only do 2.4 or 5ghz, with 5ghz i get more speed on the devices but i am forced to use 2.4 because of my esp32
would love to see support, even if its only experimental, for the c6.

damnms, 2024-11-09:
Thanks, i got tricked by this "Wifi6" and oversaw the "2.4ghz". Should be -C5

RDJunc, 2024-12-04:
Hi all.  I am not new to the ESP32 family but relatively new to WLED.  I have purchased several commercial controllers from Athom and have been relatively pleased.  I have also done two of my one designs.  Not using a development board but using the Espressif device on a board of my own design.

The first one I did was a simple Christmas tree star.  To avoid the issues of device compatability, I used the ESP32-MINI-1-N4.  As I have done in the past, I included the UART device (CP2102N) and the same transistor circuitry that Espressif uses to perform the bootloading through the USB COM port.  I even put the two EN and BOOT buttons on the board even though they aren't needed to write to the flash memory.  (See the attached.)

I ended up using esp32tool from the Command line to erase, load the bootloader, and then load WLED.  The most recent version of WLED that I was able to load was v0.13.3.  Versions v0.14.3 and v0.14.4 loaded but would not come up properly.  There were several unhandled exceptions that put the device in to endless reboots.  Also, under v0.13.3, the WiFi and LED settings are saved but not the lastest effect and color used.  Any thoughts on that?

I understand that there are some development boards that are better than others and that can cause issues with WLED even if using an ESP32.  I am not sure where my board would fall because I did basically the same thing that Espressif does on their development boards.

I very much would like to base all my future designs on the ESP32-C3.  Since I am designing the boards, I can put whatever I need to in order to ensure proper performance.  The problem is that I haven't been able to find out what exactly that is.  What should I put in any new designs with the ESP32-C3 that would keep it compatible with WLED?

Thanks very much.

[24-12-04_XMAS_Star2vB_SCH.PDF](https://github.com/user-attachments/files/18008622/24-12-04_XMAS_Star2vB_SCH.PDF)
[image]

srg74, 2024-12-04:
Regarding ESP32-MINI-1-N. Basically it’s an encased pico-d4. For this variant you have to use different bin file because pinout is slightly different than esp32 based modules.

RDJunc, 2024-12-05:
THAT'S IT!!!  Thanks very much!  Why is this not on the official release page?  Now I can move forward with my other designs based on this device.

troyhacks, 2024-12-29:
As an update on progress on the RISC-V chipsets, the ESP32-P4 is running the WLED-MM fork **very** well with Arduino-ESP32 built on IDF v5.3.2 - but it's entirely limited to network pixels at the moment. 

There's still a few rough code edges, but I have several of these (still hard to find) boards in use in the wild, including one at a venue where it's driving their Art-Net lighting installation with over 11000 pixels every night with full audio reactive effects. 

Once Makuna has some NeoPixelBus magic done for the P4 it should just snap in and we'll have local LED output as well. 

@Makuna - I'd be very happy to ship you a few ESP32-P4 boards for development, BTW. Just send me your address. ❤️

Cherish-Gww, 2024-12-30:
@troyhacks 
Is it complicated to transplant to ESP32C6?

softhack007, 2024-12-30:
> @troyhacks Is it complicated to transplant to ESP32C6?

@Cherish-Gww the -P4 port from @troyhacks is based on my work for the esp32-C6 (see a few posts up). So yes it could be ported to C6, but the result will not make you happy. Currently we have no (zero, nada, nothing, nix) usable support for led pins, not for digital leds and not for analog, either. What's working is Art-Net network output to semi-professional light control boxes.

So there is still a long way ahead of us. The message is, we're moving.

troyhacks, 2024-12-31:
> What's working is Art-Net network output to **professional/all** Art-Net control boxes.

Fixed that. I said "semi-professional" once but it actually works on all Art-Net controllers, and it's stupid fast. I was referring to "semi-professional" as more a price point.

It'll work on all of them, from using WLED itself as an Aet-Net endpoint to middle range $150 Chinese controllers all the way up to  "this costs a stupid amount of money because it's meant for stadium shows" controllers. 

To say the P4 port isn't useful isn't true - it works extremely well with Art-Net, and Art-Net is what you want when you're running huge displays anyway, as you can have essentially unlimited "pins" (outputs) with Art-Net. Parallel outputs, stupid fast, accurately clocked.

It costs more money, but using the P4 to drive a few local LEDs is kinda overkill even if it worked.

The P4 is for driving a LOT of LEDs... I've pushed over 65,000 LEDs with it via Art-Net from WLED.

Nothing on the market does this for $0 software at this rate of speed with full audio reactive effects.

It's useful - but more on the "I'm a crazy person" level. 😁

troyhacks, 2025-03-05:
[image]

[image]

[image]

Poor Akemi only runs at 11 FPS on 65025 pixels... but everything runs just fine. 383 Art-Net universes. 

[image]

2D matrix widths beyond 255 are also completely possible now too. 😄 

[image]

troyhacks, 2025-03-05:
https://github.com/user-attachments/assets/9e25b4dc-9760-4d43-a0d3-0941cfb416ec

AskAlice, 2025-05-19:
[image]

AskAlice, 2025-05-19:
> Similar for other very brandnew chips: [-C2](https://www.espressif.com/en/news/ESP32-C2), [-C6](https://www.espressif.com/en/news/ESP32-C6_Available) , [-C61](https://www.espressif.com/en/news/ESP32-C61_SoC), [ESP32-H2](https://www.espressif.com/en/news/ESP32_H2), [-H4](https://www.espressif.com/en/news/ESP32-H4), or [ESP32-P4](https://www.espressif.com/en/news/ESP32-P4), or [-C5](https://www.espressif.com/en/news/ESP32-C5).
> 
> To put it into some context - market availeability of ESP32-C6 was announced just 3 weeks ago on [Jan 6, 2023](https://www.espressif.com/en/news/ESP32-C6_Available). Knowing espressif, I bet it will take **at least 2 years** until stable development environments and open source libraries are ready for any "serious use" on -C6.
> 
> We should put it on the backburner, until stable support is availeable from the core libraries we need, for example:
> 
> * [ESP-IDF](https://github.com/espressif/esp-idf#esp-idf-release-and-soc-compatibility)
> * [platformIO espressif32](https://docs.platformio.org/en/stable/platforms/espressif32.html#espressif)
> * [arduino-esp32](https://docs.espressif.com/projects/arduino-esp32/en/latest/getting_started.html#supported-soc-s)
>   
>   * see also [ESP32-C6 support espressif/arduino-esp32#7713](https://github.com/espressif/arduino-esp32/issues/7713)
> * [NeopixelBus](https://github.com/Makuna/NeoPixelBus)
> 
> For now, we have enough troubles to make -S3/-S2/-C3 work. So don't expect support for other new models this year.
[image]

blazoncek, 2025-05-19:
@AskAlice you are welcome to implement. ;)

AskAlice, 2025-05-19:
wish  i could but i suck at c++

blazoncek, 2025-05-19:
Then be respectful and ask politely instead of posting insulting memes.

damnms, 2025-05-19:
OT: wow, if this meme is insulting for you, must be a hard life then.

troyhacks, 2025-05-20:
@AskAlice 

The first ESP32-P4 I could physically get into my hands arrived sometime in August of 2024.

I went to Burning Man and got back early September and started work and had "first lights" over Art-Net by the end of September 2024.

I shipped Makuna some P4 boards somewhere along the line and around Feb 2025 we had the first physically attached LEDs on the ESP32-P4 via NeoPixelBus.

So... within 1 month of having one in my hands it was booting WLED-MM and doing things, and within 6 months we had it "fully" working.

So that's the output of 2 people and cost me personally a few hundred dollars to get some of the earliest ESP32-P4 dev boards available - not to mention 3 solid weeks of all my free time just to get WLED to work with IDFv5 and also figure out how to build the Arduino-ESP32 framework for IDFv5 and integrate the new API needed for these boards into WLED which wasn't even *released* officially from Espressif for the P4 and the other new boards.

All this work and I'm probably the only person using WLED on the ESP32-P4 in the wild. 

But it was worth it. ❤️

influenist, 2025-05-25:
Frankly cant we put Claude to work at the implementation for C6?

troyhacks, 2025-05-25:
> Frankly can't we put Claude to work at the implementation for C6?

Go nuts and have fun! Show us what you come up with.

Honestly the C6 is a poor WLED target unless you want it for very specific things - and those specific things don't have any hookups in WLED. The single core isn't great for many LEDs and/or AudioReactive, and we don't have any GUI or backend code for Matter or ZigBee to be useful.

I've found AI to be beneficial in more minor aspects of coding in WLED - solving errors and more as a teacher helping me learn some of the more obscure C++ things I don't know. It's also done some nice reformatting and condensing for some functions I've written. 

...but I haven't seen it do any heavy lifting with Espressif code. It's just not trained to know all the IDF versions and is pretty spotty on recommendations between different versions. 

AI isn't "intelligence" - it's prediction. You need to bring the intelligence to it - and you get that intelligence from actually writing your own code and learning the Espressif ecosystem. If you have the knowledge, AI can certainly help - but without the knowledge you're just creating junk.

Even with the simple tasks I've used it for successfully it's sometimes dumb as a bag of rocks. I've had to "argue" with Claude and also point out its mistakes. "Why did you remove this line?" "Oh sorry you're right, I guess that was needed."

If I didn't (mostly) know what I was doing, the errors would just compound.

troyhacks, 2025-05-25:
The only current board that represents a full evolutionary spec jump in the ESP ecosystem is the ESP32-P4. 

Other boards have specific uses, but boards like the C6 and the C5 are mostly positioned as co-controllers for other boards.

ESP-Hosted is the glue that combines these - like using the C5 for 5Ghz WiFi with another ESP32 board. By itself it isn't really that powerful, but it's meant as an offload for 5Ghz WiFi. 

Most of the ESP32-P4 boards use a C3 as the WiFi source, for example, via ESP-Hosted. The P4 doesn't have native WiFi but it works really well with the C3 doing that function.

But then you have potentially two boards you need to flash firmware to, and the C3 on the P4 dev boards doesn't have a nice and simple USB port - so you need to get up to speed on a lot of things. It's not insanely hard, but it's also not simple.

ESP-Hosted also opens up the potential for Bluetooth and WiFi coexistence - but you need two boards for this to work.

influenist, 2025-05-25:
Thanks for the reply. I use a stack of AI tools on daily basis. Yes its not ideal in all cases and you need to be aware of the pitfalls. But sometimes it comes with good a approach - so I will give the proposed solution a try and see what comes out of it. 

Regarding your comment on the C6/P4 - I have a specific project which I want to use super mini's with. To be precise I want to make g9 bulbs smart with a supermini and addressable cob leds. According to me the c6 should be more stable with WLED than a c3 super mini would be. Aside from the newer BT and WIFI version (better range). Anyways, will give it a try on my side and provide you with feedback later on.

troyhacks, 2025-05-28:
[image]

troyhacks, 2025-08-12:
Today I managed to get the Parallel IO peripheral on the ESP32-P4 to produce very usable and extremely stable WS281x data.

The code is a bit rough but this means a "driverless" output of LED data is possible with an accelerated path.

Runs at around 124 FPS on my P4 for 256 pixels in WLED-MM-P4. Max speed is ~128 FPS. 

Clearly this can be improved but it's a solid proof of concept. 

This should also be applicable to the ESP32-C6 and the ESP32-H2, at least according to Espressif's docs. 

This uses "single pin parallel output" (what we might call serial, ha) which is a feature of the peripheral. Up to 16 pins can be used, but the bit shifting math makes my head hurt. 

There is also the Bit Scrambler peripheral which seems to be useful for this kind of task, but that's a steeper learning curve as it's a state machine with its own assembly language which you need to use to make it do things. Not sure how simple that is yet with how we build WLED. (I do have P4 assembly running inside WLED but this seems... different.)

Anyway, share and enjoy:

https://gist.github.com/troyhacks/86dfd5dd1519addcc49bea7d7957728b

schildbach, 2025-11-07:
> matter devices need a secure boot procedure

I think only for matter certification; not required for pure implementation.

netmindz, 2025-11-08:
For those interested in the work going on towards C6 and P4 then have a look at 

C6 experiments - https://github.com/MoonModules/WLED-MM/pull/138

ESP-IDF-V5 - https://github.com/MoonModules/WLED-MM/pull/262

P4 support - https://github.com/troyhacks/WLED/tree/P4_experimental

TimyIsCool, 2025-12-07:
Is there any beta firmware I could compile and flash onto my C6 at the moment?

troyhacks, 2025-12-19:
So yeah, the C5 works now too.

And with 5GHz WiFi. 😁

[image]

jonny190, 2026-02-06:
Just sent a PR with a working ESP32-C5 Build https://github.com/wled/WLED/pull/5354

djmaus73, 2026-02-08:
Hi to you all , 

testing the esp32P4 eth with  ( P4 experimental )  using 96x96 pixels ws2812 at 12 outputs  ,  most animations runs at 47 frames/s
and DDP just cant get over 10 frames/s and glitchy  
but it works ;) 
Till now GREAT JOB !!!
[image]

softhack007, 2026-02-09:
First Light with C5 5Ghz Wifi in core WLED 🚀 

There is still a long way to go, but thanks to various contribution (special thanks @jonny190 for his initial C5 port), we have something to continue improving. Currently we need a lot of hacks and bleeding-edge libraries, because -C5 is not yet supported officially by most libraries we use (like NeoPixelBus LEDs driver).

* https://github.com/wled/WLED/pull/5048

The C5 might become an interesting alternative to the -C3, not only for 5Ghz support
* It is faster than -C3 (240Mhz vs. 160Mhz)
* It supports PSRAM
* You can get it with 8MB flash

* If you want to help with development, a PR against our [``V5-C6`` branch](https://github.com/wled/WLED/tree/V5-C6) is always appreciated.

<br/>

[image]

jonny190, 2026-02-10:
@schildbach Thanks for the feedback - some really good points that have shifted my thinking on this.

> Note ESPHome tackles the flash size problem by allowing (at compile time) either Wifi or Thread, not both.

A compile-time switch (`-D WLED_USE_THREAD`) is definitely the way to go over dual-stack. Worth keeping in mind though that ESPHome's workload is quite different - it mostly pushes small sensor readings and receives simple commands. WLED has a full web UI with JS/CSS assets, WebSocket-driven real-time previews, and firmware binaries around ~1.5MB. So the bandwidth constraints will bite harder here than they do for ESPHome.

> I don't see why OTA would not work? In fact, virtually all Thread devices update their firmware over the air.

You're right, OTA would work. It'll just be noticeably slower on a ~1.5MB WLED binary at Thread speeds compared to WiFi. ESPHome handles this with `http_request.ota` which would work for WLED too. Not a dealbreaker by any means, just a user experience difference worth noting.

> Thread is IPv6, so if the bandwidth and latency is not a problem usecases should carry over without changes at protocol level.

This is honestly the strongest argument and the one that's changed my view the most. Since Thread is IPv6, AsyncWebServer, mDNS and UDP sync should technically work without protocol-level changes. WLED's web UI is ~200KB compressed so it would load slowly over Thread, and real-time WebSocket effect previews probably wouldn't be practical. But for everyday use - switching presets, adjusting brightness and colour - it should be perfectly usable.

> Wouldn't that require Matter already? Or are you thinking of a custom protocol for this?

Yeah, fair point - the border router idea was overcomplicating things without Matter as a protocol layer. Better to keep it simple.

One thing I do want to flag is that it's not quite a clean transport swap. WLED has WiFi assumptions scattered throughout the codebase - AP mode fallback, signal strength display, WiFi scanning, channel selection, BSSID locking, Improv serial provisioning. All of that would need Thread equivalents or to be conditionally compiled out. Definitely doable, just more work than it might seem at first glance.

That said, I think a realistic roadmap would be:

1. Compile-time Thread instead of WiFi, targeting C5/C6/H2
2. Basic control works over IPv6 (HTTP API, JSON, presets)
3. Web UI works but loads slowly - fine for configuration, not for real-time control
4. OTA works but is slower than WiFi
5. Provisioning via BLE or Thread commissioning instead of WiFi AP

The ESPHome OpenThread component would be a great reference for the low-level setup. The main unknown is flash budget - whether OpenThread + WLED fits in 4MB, or whether 8MB boards become a requirement. Would be worth building ESP-IDF's OpenThread example and checking the size before committing to this.

schildbach, 2026-02-10:
Regarding commissioning of new devices, you've got a point. There is probably no equivalent of an AP for Thread. You could spin a second fabric. but there are no "clients" that allow to just pick a fabric and provide credentials manually.

With ESPHome, you currently have to compile in your credentials statically, e.g. [in the form of TLVs](https://esphome.io/components/openthread/#dataset-tlv-configuration). I assume later they will support Thread in [Improv-BLE](https://www.improv-wifi.com/), but the Bluetooth stack consumes a lot of Flash/RAM. Maybe they'll implement just the Matter-Commissioning some day, then you could use the SmartHome-App of your Ecosystem (e.g. Home-Assistant).

djmaus-build, 2026-02-19:
Testing esp32c5 ( v5-c6 ), works great with only 1 led output ws2812b ? 
Setting up 4 outputs in wled , just 1 output drive the leds ( other 3 outputs no lightup at all ) 

I'm very happy with the first results 
Wifi 5ghz works formidabele with tpm2.net protocol 
It runs very stable 
Tested on 2 boards 
esp32c5 4mb ( no sdram)  esp32c5dev    build 0.16
esp32c5 ( R8n4)   sdram    esp32c5dev_qspi   build 0.16
Sdram version runs little bit more smoothly 

Hoop for a update 😁

Happy Building everybody 🥳🥳🥳

[1 more reply]