wled/WLED #3080: Interupt PIR usermod countdown

Shaps-77, 2023-02-04:
Is there a way to temporaily interupt the countdown timer function of the PIR usermod?

My situation is this.. I'm currently using the PIR and Rotary encoder usermods to control my ensuite lights. I have the PIR setup to turn on some ambient lights and the rotary encoder takes care of brightness and using its built in momentary button where 'short press' triggers preset macros and 'long press' powers off..

Here's my situation... as I enter the room the PIR triggers the lights on and the timer starts the countdown to turn off..(This is great for most situations) buttttt.. Is there a way that any input from the rotary encoder or button can suspend/cancel the PIR countdown function until WLED is powered down? Thus keeping the LEDs on in their current state.. Otherwise it seams I need to wait until the PIR counts all the way down before I can turn on the lights manually using the button or Rot Encoder... at the moment even if I use the button or encoder whilst the PIR usermod is counting down, the inputs are registered and the lights change etc.. but the countdown continues until WLED powers off.

Shaps-77, 2023-02-06:
Tested and PIR interrupt working inline with expectations above although I appear to be getting temperature sensor errors now..?   Getting constant "0 Sensor Error!" message on INFO page..
Is it just an issue with the USERMOD_DALLASTEMPERATURE and the usermod-enhance branch??

blazoncek, 2023-02-06:
@Shaps-77 for Dallas take a look at https://github.com/PaulStoffregen/OneWire/pull/93

PR #3081 (2023-02-06): Usermod enhancements
- added onStateChange() callback
- added examples & comments to usermod_v2_example.h
- PIR sensor cancels countdown on state change

Should close #3080
closed 2023-02-06 by blazoncek