wled/WLED #3083: Hue sync makes WLED unstable

ceeeekay, 2023-02-07:
### What happened?

WLED regularly restart when synced to a hue bulb. If I disable hue sync, WLED is stable.

I can literally sit in the WLED interface on the info pane and watch the uptime reset to 0 after a couple of minutes.

Disabling hue sync gives me many many hours of uptime.

Hue settings are:
Poll: 2500ms
Sync: On/Off (no colour or brightness).

This has been stable in a previous WLED version (I'm sorry I can't tell you exactly which - way back at 0.9.x I think). Previously I was running on $3 Ali ESP32 boards so I suspected these boards might be the issue, but have since fresh installed on M5Stack Atoms. and the problem persists.

Hue hub is up to date, network is stable. I can't really find anything that causes the problem other than having hue sync enabled.

### To Reproduce Bug

* Enable Hue sync with correct light ID and bridge IP.
* Watch WLED uptime

### Expected Behavior

WLED shouldn't crash/restart every couple of minutes.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b1 (build 2212222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
n/a
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-02-07:
Use debug builds from @srg74

ceeeekay, 2023-02-07:
Hi there, is this https://github.com/srg74/WLED/releases/download/v0.14.0-b1/WLED_0.14.0-b1_ESP32.bin or is there a specific debug release somewhere?

blazoncek, 2023-02-07:
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b2/Debug_builds

ceeeekay, 2023-02-07:
Ah, problem here in the sync menu:

` This firmware build does not include Philips Hue support.`

So I'm unable to reproduce with this specific debug build.

blazoncek, 2023-02-07:
Or request one from @srg74

ChuckMash, 2023-02-07:
It may be worth looking at the available APIs from HUE again.
From what I can tell, the V2 API is starting to solidify and become more robust, it already includes a text/stream call that seems like it could allow WLED to receive updates from the bridge, rather than having to poll for them.

`curl --insecure -N -H 'hue-application-key: <appkey>' -H 'Accept: text/event-stream' https://<ipaddress>/eventstream/clip/v2`

For instance, will end up returning a stream of event updates that could be parsed for desired information.

ceeeekay, 2023-02-07:
> Also this seems like it may be related to https://github.com/Aircoookie/WLED/issues/2495 https://github.com/Aircoookie/WLED/issues/2495#issuecomment-1018036263

Yep it's the same issue. Close this one, or should I keep trying to get some logs?

wusapa, 2024-01-27:
I also have the problem that my WLED keeps restarting when I sync it with my Hue lights. Are there any fixes in sight for this? 

Version: v0.14.1

closed 2024-06-03 by github-actions[bot]