wled/WLED #3084: Json API bug?

savvakis1, 2023-02-07:
### What happened?

Hello to everyone,

I recently got a ws2812 led matrix (16x16) and I am trying to sent JSON API commands in order to set each led to a specific color to form an image. Well the problem (at least for me) is that sending the command in rgb format/values (e.g. [255,255,0] etc.) I get a : typeerror: cannot set properties of undefined (setting 'on').

With a quick search, this error has to do with Javascript (?)

After trying to send the command in hex color values (e.g. "FFFF00" etc.), all seems to work perfectly. 

### To Reproduce Bug

Trying to set a fairly amount of leds (almost half of them) with rgb values I get this type of error. 

### Expected Behavior

This is also happening while selecting "peak" to check visually the result.

Tldr; with rgb values: error but with hex values: works perfectly.

I hope this helps.

### Install Method

Binary from WLED.me

### What version of WLED?

Wled 14.0 beta

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2023-02-07:
Hi and thank you for the report! Can you please share the exact command you send that caused the issue?

Pretty sure it is caused by a JSON buffer overflow, as the array format for colors is quite inefficient. The solution is to send multiple commands, updating only part of the LEDs in each and preferring hex syntax.

dosipod, 2023-02-08:
I think i seen issues like that on esp8266 only . Below is 16x16 pixel art which works without any issue on esp32

blazoncek, 2023-02-08:
There is only a limited amount of RAM available and sending huge number of LEDs in a single API call will overflow the buffer (actually it won't but will still produce error),

Split your API call into multiple calls as a workaround. Another workaround (to increase the number of LEDs in a single call) is to use hex notation.

Even though this may seem bug it is not. It is a platform restriction imposed by hardware.

closed 2023-02-08 by blazoncek

savvakis1, 2023-02-08:
Thanks for helping out, I guess I'll settle with hex values commands. I've written a python script do do the translation from rgb to hex. 

Again, thank you very much!