wled/WLED #3086: Will this board work?

TysTies, 2023-02-08:
### What happened?

I Tried to make an esp32 c3 by seed studio work, but it does this weird thing where the esp flasher just scrolls so fast and you can't read what it is saying. here's the board: https://www.amazon.com/dp/B0B94JZ2YF

### To Reproduce Bug

Buy that board, (it might be unavailable but check again in a few days and it will reappear) Then try to Flash wled

### Expected Behavior

already did in the first box/  n/a

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0-b1_ESP32.bin

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-02-08:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2023-02-08 by blazoncek

TysTies, 2023-02-08:
I can't. Please just respond to me. I am on a school computer. And they
don't allow any websites on request and I don't have any other devices at
home other than this school computer.

blazoncek, 2023-02-08:
Yon need a binary for C3.
C3 is still experimental and not supported.

BTW ask your teacher to help you. He/she will understand if you are doing legitimate work.

TysTies, 2023-02-08:
I am not in a class that does stuff with esp boards. i do that on my own
time at home. thanks for responding, could you send me a link for the
binary?

blazoncek, 2023-02-08:
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b2/S2_S3_C3