wled/WLED #3087: NeoPixelBrightnessBus being deprecated

softhack007, 2023-02-10:
**Is your feature request related to a problem? Please describe.**
I just saw the announcement from makuna
* https://github.com/Makuna/NeoPixelBus/discussions/644

**Describe the solution you'd like**
We are currently using NeoPixelBrightnessBus objects in bus_manager.h.

It seems that there is a replacement with similar functionality, however considering that we also have challenges to related to gamma correction LUT and "non-natural dimming" (like https://github.com/Aircoookie/WLED/issues/2767), destructive brightness and color values < 28 being mapped to 0 due to gamma LUT (several discussions on discord) ,  and we want to support higher bitdepht on PWM(https://github.com/Aircoookie/WLED/issues/2868),...

 Well it might be a good time to plan for a redesign of bus_manager.h, and replace NeoPixelBrightnessBus with something more appropriate.

Thank you for your ideas for making WLED better!

blazoncek, 2023-02-10:
It should not be that difficult, though tedious, as entire WLED interacts with bus manager and not bus wrapper.
Btw bus_manager is an abstraction layer providing API (actually replicating NPBB from bus_wrapper).

blazoncek, 2023-02-13:
@softhack007 will you have time to test the above commit?

softhack007, 2023-02-13:
> @softhack007 will you have time to test the above commit?

Not this week, sorry. I'm away from home until 20th.
But I'll surely take a look and test next week.

Aircoookie, 2023-02-14:
Color recovery (although lossy) is quite trivial: `recovered = (original*256) / (brightness+1)`. I would therefore propose we use the standard `NeoPixelBus` instead of `NeoPixelBusLg` and handle the brightness scaling in bus manager. This also gives greater control for >8 bit busses. As for the new gamma correction capability (which is cool to have in the library now!), it would likely not make sense to delegate it to NPB, as we still need it within WLED for e.g. analog busses and for functionality like having correction enabled for FX but not for live data.

@blazoncek I will do ESP8266 testing of #3091 on Wednesday, will test adding brightness scaling to get/setPixelColor :)

blazoncek, 2023-02-14:
NPBLg will return unmodified (original) pixel value from SetPixelColor(), there should be no need to *recover* old value from it.
At least that was my impression after a glimpse of the NPBLg code.

There are some clever tricks in NPBLg for 16bit LEDs.

Aircoookie, 2023-02-14:
As far as I understand, it returns the original (as in, already scaled) value as written to the LED. So if you set 255,128,0 at 50% bri it would just return 127,64,0 rather than attempting to restore the color: https://github.com/Makuna/NeoPixelBus/blob/master/src/NeoPixelBusLg.h#L92-L92

Non-lossy reconstruction is not even possible without an extra buffer, but I will give you the benefit of the doubt and first test the PR as is.

blazoncek, 2023-02-14:
Looks like my brief read-through was not accurate enough.
And BTW, I did not touch post-processing in bus_wrapper. I merely replaced NeoPixelBrightnessBus with NeoPixelBusLg. Nothing more.

commit 9434d17e (2023-02-12): Replace NeoPixelBrightnessBus with NeoPixelBusLg
Resolves #3087
commit e0011655 (2023-04-27): Add Ucs890x support and swaps NeoPixelBrightnessBus with NeoPixelBusLg (#3091)

* Add UCS890x support.

* Fixes

* Update NeoPixelBus to 2.7.3 for UCS8904 support.
Update ESP8266 core to 4.1.0

* ESP8266 compile fixes.
- use PlatformIO framework and toolchain
- add compiler warning suppression
- remove IRAM_ATTR to fit in IRAM

* Replace NeoPixelBrightnessBus with NeoPixelBusLg
Resolves #3087

* Update to NPB 2.7.4

* Internal NPB color conversions.

* Fix errors due to merge with SPI Hz methods

Regenerate settings page HTML

---------
closed 2023-04-27 by Aircoookie