wled/WLED #3093: Police effect removed?

jeroen85, 2023-02-12:
### What happened?

I have my WLED connected to Home Assistant. Some automations make use of the `Police` effect (in combination with my audio setup playing Fireman Sam ;-) , my son loved it)

With the current version WLED 0.14.0-b1 "Hoshi", I am not able to use the police effect anymore.
It is removed? I don't see any 'Breaking Changes' within the release notes.

Can it be restored somehow?

### To Reproduce Bug

Run HA service
```
      - service: light.turn_on
        target:
          entity_id: light.ledstrip_tv
        data:
          effect: Police
          brightness: 255
```

### Expected Behavior

Enjoy the red/blue police light effect.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fishbone-git, 2023-02-12:
Use two dots effect with red/blue at full intensity, then save the preset.  Many of the reused effects where just the colours are changed are either gone, or going away soon.  The presets can handle all of that and are just as easy to call from an HA service.

blazoncek, 2023-02-12:
Yes, it was removed and it will not be restored.
List of replacements can be found in [KB](https://kno.wled.ge)

closed 2023-02-12 by blazoncek

jeroen85, 2023-02-13:
Ok, thanks. 
_A "breaking change" warning in the release notes would have been appreciated in case effects are being removed._