wled/WLED #3094: SK6812 RGBW swap not working

sammers230, 2023-02-13:
### What happened?

Firstly a quick thanks for this project, got a few bits setup and working like a charm. 

Down to the nitty gritty..

I’ve recently presented some more athom WLED controllers and mistakenly ordered some WS2814 strip. I’ve reflashed the athom with the latest beta version 0.14.0-b1 after compiling it for the Athoms ESP8285 chip. 

Going into settings as usual and the option to swap the white and a selected colour channel shows bit when picking to swap W&B as I need to and saving, it seems that it refuses to save and instead the UI just goes back to showing none and never actually updating the desired channel swap. 

### To Reproduce Bug

With WS2814 led strip the SK6812 RGBW option is required and the colour order of BRG to match the WS2814s order for control. 
selecting the option to swap W&B and saving results in the setting not saving or updating in any way and just stays at none for the swap option. 

### Expected Behavior

Blue and white channels of the LED strip should be swapped on the control side so the desired colour patterns are correct. 

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1 (2212222)

### Which microcontroller/board are you seeing the problem on?

ESP8266, Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sammers230, 2023-02-13:
After a little more playing around. It seems if I add another set of leds on another gpio (can’t physically add them but just in led preferences) and telling both to swap colours it has bumped the firmware into accepting the swap options 

Incase anyone else has the same WS2814 LED strip and is having an issue setting up I have 

Colour Order: BRG 
Swap: W&G

To confirm I also restated my controller and the settings have indeed saved.

I’ll be updating the other Athom controller I have tomorrow to the latest version 0.14.0-b2 also and will check to see if the issue of not saving the colour swap presents itself again before looking to close this.

blazoncek, 2023-02-13:
This was brought up a few times now but unfortunately I cannot reproduce.
As I stated elsewhere (#2646) such behaviour will require debug build to diagnose and monitoring serial output.

What I suspect is uncommon GPIO allocation so a cfg.json file and/or LED settings page screenshot will help diagnostics.

Carl-Wilhelm, 2023-02-23:
> After a little more playing around. It seems if I add another set of leds on another gpio (can’t physically add them but just in led preferences) and telling both to swap colours it has bumped the [...]
I have the exact same issue. Using a NodeMCU. Was able to work around the problem the way that sammers230 described.

mrosseel, 2023-02-27:
exact same issue, using a Dig Quad Uno (esp32+ethernet hat). The workaround described did indeed fix the bug, although I had to switch from using led1 to led2.

mrosseel, 2023-02-28:
update: got it working, the reason the swap was not saved: because of various errors in the LED setup (such as entering the strip having 300 leds, but they are grouped per 6 so the number to be entered was 50.). My experience: if swap is not saving, first correct other strip settings. I could have known the setup was wrong by the wonky animations.

cthunberg, 2023-03-08:
I had the same issue with a Dig Quad but I was able to correct it through a factory reset and a reflash of the firmware. I did not restore a configuration.

closed 2023-03-28 by blazoncek

astraliens, 2023-04-27:
got the same issue with esp8266 witty cloud and v0.14.0-b1 firmware installed from https://install.wled.me
when "Off Refresh" checkbox is checked we've got "Swap" option always reset to "None" after saving config, but if "Off Refresh" is unchecked - "Swap" is saving without problem

yannickdm, 2023-05-11:
Same issue over here. Using the QuinLED-Dig-Uno, with the WLED firmware on it. Currently using 0.14
Unable to save the swap when Off Refresh is checked. Even after a reboot the setting MIGHT be gone.

glbarnes67, 2023-11-20:
Same issue with Swap setting not saving with my ESP8266 D1 Mini and BTF-Lighting WS2814 RGBW (RGB-Cool White).  Unchecking "Off Refresh" finally allowed me to save the Swap setting.  Even though I needed to swap White with Blue, Selecting W & G gave me the desired outcome.

5weetbob, 2025-01-20:
When configuring the ws6814 in wled, make sure you uncheck "None" at the bottom when it asks "Auto-calculate W channel from RGB:"

None means do not use white led, all other choices will use the white leds.