wled/WLED #3097: Load last preset on power up

yasinkryk, 2023-02-15:
We installed an LED light in an area with a lot of wifi signals,

We gathered all the leds on the access point device using 16 ESP8266 modules.

but after 15-20 minutes after choosing one of the presets we have prepared, some modules time out and return to the start animation.

Question:
Is there a feature such as remember the last state when the power is turned on and off?

blazoncek, 2023-02-15:
No. Best approximation is Auto Save usermod.
But that does not do presets, only manually modified WLED state.

yasinkryk, 2023-02-15:
thank you, I haven't come across a topic about how to Auto Save usermod.

softhack007, 2023-02-15:
It's here https://github.com/Aircoookie/WLED/tree/main/usermods/usermod_v2_auto_save

closed 2025-06-26 by blazoncek