wled/WLED #3098: Seven Segment Display Reloaded usermod causing crash/reboot/lockups

smartroad, 2023-02-15:
### What happened?

I am using the "[seven_segment_display_reloaded](https://github.com/Aircoookie/WLED/tree/main/usermods/seven_segment_display_reloaded)" but it seems to cause the install to crash/lockup

The log below has a start crash not long after turning on, then a crash when switching preset, the another crash while running after starting after the preset crash.

### To Reproduce Bug

I compiled with the usermod but when running it the WLED it will crash

### Expected Behavior

No crash? LOL (Sorry not trying to be sarcastic! Just don't know what else to say 😁)

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b2 (build 2302120)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
18:34:32.442 -> 
18:34:32.442 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
18:34:32.442 -> 
18:34:32.442 -> Exception (29):
18:34:32.442 -> epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
18:34:32.442 -> 
18:34:32.442 -> >>>stack>>>
18:34:32.442 -> 
18:34:32.442 -> ctx: sys
18:34:32.442 -> sp: 3fffec10 end: 3fffffb0 offset: 0190
18:34:32.442 -> 3fffeda0:  00000005 3ffeee88 00000002 40101670  
18:34:32.442 -> 3fffedb0:  4027fa6b 3fff667c 3ffeabbb 4027fa00  
18:34:32.488 -> 3fffedc0:  00000002 4027f9a7 00000002 4027eb00  
18:34:32.488 -> 3fffedd0:  4027eb29 3fffee80 3ffeee88 00000016  
18:34:32.488 -> 3fffede0:  4027c598 3fffee80 3ffeed34 3ffee738  
18:34:32.488 -> 3fffedf0:  3ffeb7c4 3fffee80 3fffee80 4025da20  
18:34:32.488 -> 3fffee00:  656e614a 4bc6a700 220c49ba 00000000  
18:34:32.488 -> 3fffee10:  00000000 00000000 4bc6a7f0 00000000  
18:34:32.488 -> 3fffee20:  4027890d 00010000 401013d8 0003c752  
18:34:32.488 -> 3fffee30:  ffffffb7 3ffee848 3ffeb7d4 3ffeee88  
18:34:32.488 -> 3fffee40:  3ffedc50 00000017 00000000 4027d28f  
18:34:32.488 -> 3fffee50:  00000000 3fff667c ffffffb7 00000000  
18:34:32.488 -> 3fffee60:  00000000 3ffeee88 00000001 00000001  
18:34:32.536 -> 3fffee70:  00000000 00000000 00000020 40101058  
18:34:32.536 -> 3fffee80:  00000000 10110b0b 00640104 00000043  
18:34:32.536 -> 3fffee90:  3ffeb7e8 00000105 3ffeb7fb 3ffeb7dc  
18:34:32.536 -> 3fffeea0:  00000000 3ffeb7e8 3ffeb7ee 3ffeb804  
18:34:32.536 -> 3fffeeb0:  00000000 00000000 3ffeb80a 3ffeb8d3  
18:34:32.536 -> 3fffeec0:  3ffeb827 3ffeb843 00000000 3ffeb875  
18:34:32.536 -> 3fffeed0:  00000000 00000000 00000017 00000000  
18:34:32.536 -> 3fffeee0:  3fff6a44 4027cd02 3ffedc50 3fff667c  
18:34:32.536 -> 3fffeef0:  00000000 3ffeee88 3ffedc50 3ffeb7c4  
18:34:32.536 -> 3fffef00:  3ffeb7c4 0000012d 00000000 00000017  
18:34:32.583 -> 3fffef10:  00000000 3ffeb7ce 4028647f 3ffedc50  
18:34:32.583 -> 3fffef20:  3ffeb7b8 3fffdcc0 3ffea308 3ffea308  
18:34:32.583 -> 3fffef30:  00000080 3ffedc50 00000000 3ffe8874  
18:34:32.583 -> 3fffef40:  40285d3b 3fffdab0 00000000 40264c98  
18:34:32.583 -> 3fffef50:  3ffea308 40000f49 3fffdab0 40000f49  
18:34:32.583 -> 3fffef60:  40000e19 00000005 000c65c8 00000000  
18:34:32.583 -> 3fffef70:  00000000 aa55aa55 000000ed 40105fe5  
18:34:32.583 -> 3fffef80:  40105feb 000c65c8 00000000 732f6567  
[... 602 lines cut ...]
18:45:02.560 -> 3fffff60:  3fffdad0 3ffefbff 3fff25ae 40246aaa  
18:45:02.560 -> 3fffff70:  00000000 feefeffe 00efeffe feefeffe  
18:45:02.560 -> 3fffff80:  00000000 00000000 00000001 3fff3078  
18:45:02.560 -> 3fffff90:  3fffdad0 00000000 3fff3038 40247004  
18:45:02.560 -> 3fffffa0:  feefeffe feefeffe feefeffe 40264e3c  
18:45:02.560 -> <<<stack<<<
18:45:02.560 -> 
18:45:02.560 -> last failed alloc call: 4025489E(2261)
18:45:02.607 -> 
18:45:02.607 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
18:45:02.607 -> 
18:45:02.607 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
18:45:02.607 -> 
18:45:02.607 -> load 0x4010f000, len 3584, room 16 
18:45:02.607 -> tail 0
18:45:02.607 -> chksum 0xb0
18:45:02.607 -> csum 0xb0
18:45:02.607 -> v3969889e
18:45:02.607 -> ~ld
18:45:02.796 -> Ada
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-02-15:
Please contact usermod creator.
You will find him/her in commit history.

softhack007, 2023-02-15:
are you referring to this PR?
* https://github.com/Aircoookie/WLED/pull/3015

If yes, please keep in mind that we cannot provide support for 3rd party code (usermods), especially when the code you try to use has not passed code reviews, and it was not merge into WLED. 

If my assumption is right, then please contact the PR author, and close this problem report.

smartroad, 2023-02-18:
I'll close this then, would it be an idea to remove it from the repository as it doesn't seem to work (properly anyway)? I note that someone else says it doesn't seem to be working on an ESP8266 on the PR.

closed 2023-02-18 by smartroad

blazoncek, 2023-02-18:
If you can get it to work, make a PR and it will be updated.