wled/WLED #3101: User/Expert/Developer mode on the settings pages

Bebick, 2023-02-19:
Hello all,

I think the software is great, but I don’t always want to have all settings displayed. For me it is clearer if some settings are hidden.
Therefore I implemented a query to check in which user mode I am.

There are 3 user modes: User / Expert and Developer.

The two modes user and expert can be displayed via drop down menu. The expert mode is only activated after entering a password. In the expert mode, further functions are then displayed.

The Developer mode is activated at the bottom via a checkbox and another password request. If this mode is activated, all functions of the current page are displayed.

Now I would like to transfer these 3 user levels to other websites and use these functions there as well. Currently the code is only used on the Settings_UI page.

Do you have an idea how I can realize this?

If you see further optimizations of this code, please let me know :).

```
<!DOCTYPE html>
<html>
<head lang="en">
	<meta charset="utf-8">
	<meta name="viewport" content="width=500">
	<meta content="width=device-width, initial-scale=1.0, maximum-scale=1.0, user-scalable=no" name="viewport"/>
	<title>UI Settings</title>
	<script>
	var d = document;
	var loc = false, locip;
	var initial_ds, initial_st, initial_su;
	var sett = null;
	var l = {
		"comp":{
			"labels":"Show button labels",
			"colors":{
				"LABEL":"Color selection methods",
				"picker": "Color Wheel",
				"rgb": "RGB sliders",
				"quick": "Quick color selectors",
				"hex": "HEX color input"
				},
			"pcmbot": "Show bottom tab bar in PC mode",
			"pid": "Show preset IDs",
			"seglen": "Set segment length instead of stop LED",
			"segpwr": "Hide segment power &amp; brightness",
			"segexp" : "Always expand first segment",
			"css": "Enable custom CSS",
			"hdays": "Enable custom Holidays list"
		},
		"theme":{
			"alpha": {
				"bg":"Background opacity",
				"tab":"Button opacity"
			},
			"bg":{
				"url":"BG image URL",
				"random":"Random BG image"
			},
			"color":{
				"bg":"BG HEX color"
			}
		}
	};
	function gId(s) { return d.getElementById(s); }
	function toggle(el) { gId(el).classList.toggle("hide"); gId('No'+el).classList.toggle("hide"); }
	function isObject(item) {
		return (item && typeof item === 'object' && !Array.isArray(item));
	}
	function set(path, obj, val) {
		var tar = obj;
		var pList = path.split('_');
[... 316 lines cut ...]
				themeOptionsDiv.style.display = "none";
				themeOptionsDiv_2.style.display= "none";
			} else if (userLevel == "expert") {
				var password = '';
				var dialog = document.createElement("dialog");
				dialog.innerHTML = "<label>Expert Password:</label><input type='password' id='password_input'/><button id='password_button'>Submit</button>";
				document.body.appendChild(dialog);
				document.getElementById('password_button').addEventListener('click', function() {
				password = document.getElementById('password_input').value;
				if (password === "expert") {
					holidaysDiv.style.display = "block";
					themeOptionsDiv.style.display = "none";
					themeOptionsDiv_2.style.display = "block";
					dialog.close();
				} else {
					alert("False Password. Please do it again.");
				}
				});
				dialog.showModal();
			}
			}
			</script>
	</body>
</html>
```

blazoncek, 2023-02-19:
Please fork and make a [PR](https://github.com/Aircoookie/WLED/wiki/How-to-properly-submit-a-PR). It is easier to see what you mean that way.

Bebick, 2023-02-19:
Okay I did that.
My code is here: [https://github.com/Bebick/WLED-main-4](url)

softhack007, 2023-02-19:
I am not sure that what you try to do is a good idea. Hiding away parts of the settings pages is going to cause lots of confusion. It might give us a load of support requests if that code becomes part of WLED. 

* WLED is written assuming that all settings will be availeable for users. Removing some pages may lead to very unexpected consequences.

* I'm not sure why you want to hide away usermod setup? If people build WLED with usermods enabled, why should usermod setting be hidden?
* please do not include a plain text password inside the JS parts. JS is delivered to the browser, and anybody who knows how to start IE developer mode will be able to retrieve it.
* what "other web pages" would you like to split up? And why?
* I don't understand the motivation -  can you describe a realistic use case where several user levels would be needed?

Aircoookie, 2023-02-20:
With settings, my stance is always "if you don't know what it does, leave it be", as such personally I see settings "advancedness" modes as an overhead and always set them to the highest level in every program even if I am just a novice user.

Given that not everyone might share this opinion though and many settings may overwhelm some users, I'd be open to include a Basic/Advanced toggle in the upcoming settings rework. I don't see any reason for a developer mode either though, "developer-only" settings are realized with compile flags :)
And I just have no idea why one would password-protect advanced settings mode, sorry.

Bebick, 2023-02-25:
First of all, thank you for your opinion.

So I would also say that the developer mode is not absolutely necessary.

For me, the motivation behind it was that there are perhaps "too many" settings for some users, which you might need once a year and can therefore also hide. And the whole realized via the user and expert mode.
It would be best if this mode is usable for multiple pages - so to set a kind of variable when the mode is on and this variable you can then arbitrarily in the sections you want to have hidden. So everyone can still customize the pages individually and it does not have to be adapted the complete documentation. But only a short explanation how to deal with this variable.