wled/WLED #3103: Playlist repeat will not save at repeat:1; always goes to repeat:2

alvindotdev, 2023-02-24:
### What happened?

I wrote this in a prior closed issue where someone else posted a similar issue; but figured i should open it anew

I'm seeing the same issue as last published. 

1. I created a playlist containing several presets
2. I do not want it to repeat indefinitely - i do not want it to repeat at all. so i uncheck that box and set repeat to 1 time
3. I save
4. Immediately the playlist runs on my led strip without repeating
5. If i now expand the playlist again to see the settings, repeat is set to (2)
6. If I view the api call, i can see repeat:2 in the json
7. If i play the playlist anytime going forward, it will repeat twice

I can't seem to permanently make the playlist stay at repeat 1. Even if i update the API call manually to set repeat:1, then save, it will still do the same steps 4-7 above.

This occurs in both the iphone app and the web browser mode

Setting repeat to (1)
[image]

Save and it is not expanded any longer -- it will run the playlist automatically and not repeat which is GOOD
[image]

Any subsequent play to the playlist will repeat (2) times, and when i expand to edit, (2) will be written in the playlist repeat setting
[image]

[image]

Even if i edit the api command directly to repeat:1...
[image]

It will do the same things described above --run without repeat successfully 1 time , and subsequent runs will repeat:2.

[image]

I just flashed wled 1 day ago on the latest version on a ESP32: HiLetgo ESP-WROOM-32 ESP32 ESP-32S Development Board 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Processor Integrated with Antenna RF AMP Filter AP STA for Arduino IDE

### To Reproduce Bug

1. I created a playlist containing several presets
2. I do not want it to repeat indefinitely - i do not want it to repeat at all. so i uncheck that box and set repeat to 1 time
3. I save
4. Immediately the playlist runs on my led strip without repeating
5. If i now expand the playlist again to see the settings, repeat is set to (2)
6. If I view the api call, i can see repeat:2 in the json
7. If i play the playlist anytime going forward, it will repeat twice

### Expected Behavior

Repeat:1 should be saved and remain going forward for the playlist

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.0-b1 "Hoshi" Build 2212222

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-02-24:
This is fixed on latest , make sure to compile from source 
[image]

closed 2023-02-24 by blazoncek

alvindotdev, 2023-02-24:
Impressive speed, thank you. I will take a look