wled/WLED #3104: Add support for Intertek (American Lighting / Better Homes and Gardens) outdoor pixels

forkineye, 2023-02-24:
I'm tracking this for my firmware as well here - https://github.com/forkineye/ESPixelStick/issues/622.  Once I get it sorted out, I can work on a PR for WLED unless someone else wants to do it :)

This is to add support for 24V Intertek outdoor flex lights which are sold by Walmart and Costco under their own brand names.  So far, I have come across two versions of these lights. Both follow the ws2811 protocol very closely, but with padding between each 24bit chunk of pixel data and very long reset / latch times.  For the newer models of these lights, the padding and reset is much larger and seems to be required.  The older models of these lights will work with standard ws2811 timing, however the newer ones will not.  [Here](https://github.com/forkineye/ESPixelStick/files/10825728/New_Intertek.zip) are the DSview .dsl and exported .vcd files from a capture of the newer model controller.  The pixel bit timing is the same as ws2811, however the following discrepancies of the newer lights were noticed:
Inter-pixel padding of 20us pulled low
Reset / Latch time of 12.5ms

Newer models that do not work with ws2811:
https://www.walmart.com/ip/Better-Homes-Gardens-Color-Changing-Chasing-Neon-Flex-Light-16-4-feet/752676062
https://www.costco.com/ProductDisplay?catalogId=10701&partNumber=100788981

At the time of purchase (2/2023), this reseller had the older model which works with ws2811:
https://www.amazon.com/gp/product/B097HY6NF3

Cheers!
-shelby

blazoncek, 2023-02-24:
Perhaps we can ask @Makuna if he ever encontered such LEDs, as he is the one providing LED driver for WLED.

Makuna, 2023-02-24:
Nothing like that yet.

Original WS211 is half that speed.  The newer ones are basically the same as Ws2812x.  This is inline with normal 800kbs speeds of Ws2812x and they have a 300us reset time.

looking at your capture, 
19.25us low signal between RGB pixels, 
12.47ms reset.

I wonder if the reset is really just a decision on the module writer for how often they update rather than a hard requirement.  The same could be said for the time between the pixels that the chips are just flexible to support it and the module abuses it.  Experimentation would be required.

I would try upping the reset time to the 12480us and see if it just works.  
If it does work, it proves the period between pixels is optional.  Then slowly lower the reset by 1000 and see where it breaks.
If it doesn't work, then a little encoding change would be required.

forkineye, 2023-02-24:
@Makuna I tried upping the reset time in the espixelstick code and no joy. I'll try doing it in wled next week when I get back.

Makuna, 2023-02-25:
Its subtle, but there is also a larger pulse between the elements of the colors, its 0.2us extra low value ever 8 data bits.

bigredfrog, 2023-02-26:
I have working and non work light strip instances from American Lighting to hand. I have an ESPixelStick3.0 and WLED VSCode based build environment that I can build / deploy to. Can anyone point me at where these timings are in the code, I can at least experiment and help with discovery a little. I have been spelunking, but not jumping out at me yet.

Makuna, 2023-02-26:
@bigredfrog What platform do you run it on?  Esp32/Esp8266/other?

bigredfrog, 2023-02-26:
I have an espixelstick3 to test on so esp8266_2m target as I understand it.

Makuna, 2023-02-26:
Assuming you are using Ws2812x in your config, find this in your WLED source (NeoEsp8266DmaMethod.h)
```
class NeoEsp8266DmaSpeed800KbpsBase
```
for the lines below it, change to match this... 
```
class NeoEsp8266DmaSpeed800KbpsBase : public NeoEsp8266DmaSpeedBase
{
public:
    const static uint32_t I2sClockDivisor = 5; // 0-63
    const static uint32_t I2sBaseClockDivisor = 10; // 0-63
    const static uint32_t ByteSendTimeUs = 10; // us it takes to send a single pixel element at 800khz speed
};

class NeoEsp8266DmaSpeedWs2812x : public NeoEsp8266DmaSpeed800KbpsBase
{
public:
    const static uint32_t ResetTimeUs = 12480;
};
```

bigredfrog, 2023-02-26:
Will do, AFK for an hour or so.

bigredfrog, 2023-02-26:
At desk

When I look for NeoEsp8266DmaMethod.h or NeoEsp8266DmaSpeed800KbpsBase

I am only finding that in .pio\libdeps\... folders, rather than source searching in the Platformio / WLED source tree.

Am I missing the obvious somewhere? 

Also happy to do this by discord if it helps...

bigredfrog, 2023-02-26:
Is this a dependancy pulled in at build time? Can I safely edit it in place without it getting flushed? Trying anyway

bigredfrog, 2023-02-26:
Well thats odd, following the same process as prior, I now get update failed when trying to OTA the esp8266_2m.bin file 794KB

bigredfrog, 2023-02-26:
It does not want to take an OTA of the official release either, so assuming something is bad in origin/main

Going to manually program back to a known 13.x and then OTA this build again

Makuna, 2023-02-26:
I am not familiar with WLED build details so I can't help there.

bigredfrog, 2023-02-26:
Baselining back to 13.x with USB cable, then OTA to my hand build is again successful

Good strip looks fine

Bad strip is still a flickery mess

I cannot however convince myself that the changes in NeoEsp8226DmaSpeedBase are inherited as they come from dependencies that may be cached. I will try and break the build so see if it fails.

bigredfrog, 2023-02-26:
Although making a bad edit to NeoEsp8226DmaMethod.h does break the build. Fixing the edit allowed build to complete, but did not update time stamp on the esp8266_2m.bin, so very suspicious that build toolchain has layers of caching that I cannot have confidence have inherited changes.

Tried changing values in header file semi random, and I cannot seem to get a deterministic new binary image with updated time stamp. There is something I am not understanding here.

bigredfrog, 2023-02-26:
I have tried rebuilding while grossly changing the constants proposed, but from what I can work out the checksum on the generated binary is not changing, therefore I have no trust that my changes are taking effect.

Appreciated @Makuna the WLED VSCode build process may be opaque to yourself as well.

Hopefully @forkineye is more empowered!

bigredfrog, 2023-02-26:
rambling further in https://discord.com/channels/789177382221119519/789177382221119524/1079475721581367347

To keep this a bit cleaner in here

bigredfrog, 2023-02-26:
If as I now believe I have that longer reset time working and it has not helped, what does "If it doesn't work, then a little encoding change would be required." refer to?

Makuna, 2023-02-26:
At a low level in my library, it converts (encodes) the pixel data into a bit stream that the ESP hardware peripheral consumes to produce the needed output on a pin.  It has to be enhanced to support interpixel delays into that data stream.
Create an issue on github (makuna/neopixelbus) and reference this issue.  
Now what platforms is it required for, ESP8266 and ESP32?

blazoncek, 2023-02-26:
WLED only runs on ESP8266 and ESP32 (some variants of 32 in experimental but working phase).

Makuna, 2023-02-27:
I have a branch published [DelayBetweenPixels](https://github.com/Makuna/NeoPixelBus/tree/DelayBetweenPixels) that contains changes for Esp8266 to support a new method NeoIntertekMethod (aka NeoEsp8266DmaIntertekMethod) that adds the support matching the capture above.

I would love to see someone try this and if it works then try reducing the reset delay (ResetTimeUs) expressed in this class to a minimum that doesn't work (binary search, divide value by two until it fails, then divide difference between failed and last worked until you find the working minimum)

```
class NeoEsp8266DmaSpeedIntertek : public NeoEsp8266DmaSpeedBase
{
public:
    const static uint32_t I2sClockDivisor = 5; // 0-63
    const static uint32_t I2sBaseClockDivisor = 10; // 0-63
    const static uint32_t ByteSendTimeUs = 10; // us it takes to send a single pixel element
    const static uint32_t ResetTimeUs = 12470;
    const static uint32_t InterPixelTimeUs = 20;
};
```

less interesting but may also be useful is doing the same with InterPixelTimeUs but just reduce it to "10". as it increments only by ByteSendTimeUs.

bigredfrog, 2023-02-27:
Will try, hopefully today. Still worried if my WLED build is incremental to pick up, but worst case if I don't see a modal change with some deep cuts of build assets, will try some intentional breaks. Not sure what my binary chop cycle time will be from built to deployed, but will advise as I get insight.

blazoncek, 2023-02-27:
@bigredfrog just use something like:

```
lib_deps = ${esp32.lib_deps}
  https://github.com/Makuna/NeoPixelBus.git#DelayBetweenPixels
  ...
```

in your platformio.ini (or override)

bigredfrog, 2023-02-27:
Understanding I am inexperienced in platfomio / vscode in general ( though a long tail in SW engineering ) is the syntax above correct to imply a branch pull, that # looks like a comment syntax to me. Is it intended to explicitly pull that branch in?

I am digging around for relevant docs now

blazoncek, 2023-02-27:
https://docs.platformio.org/en/latest/core/userguide/lib/cmd_install.html#cmd-lib-install

bigredfrog, 2023-02-27:
Thanks for that, understand the syntax now. 

I tried the build with that line in lib_deps, but it goes bang. a forest of warnings, and 10 errors, first of which is 

In file included from .pio\libdeps\esp8266_2m\NeoPixelBus\src/NeoPixelBus.h:89:0,
                 from .pio\libdeps\esp8266_2m\NeoPixelBus\src\internal\NeoGamma.cpp:28:
.pio\libdeps\esp8266_2m\NeoPixelBus\src/internal/NeoEsp8266DmaMethod.h:215:33: warning: 'maybe_unused' attribute directive ignored [-Wattributes]
         [[maybe_unused]] size_t sizePixel)
                                 ^
.pio\libdeps\esp8266_2m\NeoPixelBus\src/internal/NeoEsp8266DmaMethod.h:293:29: error: expected unqualified-id before '[' token

Going to see if I can confirm I have source from the branch manually first

blazoncek, 2023-02-27:
See the #3091 for change in `platform` needed for correct compile.
You will need newer compiler which is not included with "Tasmota" platform ESP8266 is using ATM.

bigredfrog, 2023-02-27:
Moved conversation to https://discord.com/channels/789177382221119519/789177382221119524/1079870580431523870

Makuna, 2023-03-01:
UART methods on ESP8266 cannot support the required protocol due to limitations in hardware.

The issue is that the silent part between pixels cannot be done with the UART, as there is no way to suppress the start bit that UART coms rely on.  So, every byte sent, a positive pulse will be sent at the front that cannot be suppressed, making it impossible to support the silent block between the pixels with this method.

Note that the current uart support uses this unsuppressed start bit as part of the normal pulse train.

Makuna, 2023-03-05:
That working branch updated to now support bitbang: NeoEsp8266BitBangIntertekMethod & NeoEsp32BitBangIntertekMethod.  Bitbang also now works on ESP32C3.

bigredfrog, 2023-03-06:
Latest testing implies this is not timing related. Running various approaches including bit bang on GPIO2 does not change behavior.

However, suspicion is now on the data line voltage. 

Measuring original controller it can be seen at 5V. 
On a pixel stick output can be seen at 4V
Far end data line pass on is at 6V

Next step is likely to try a level shifter on the input side.

forkineye, 2023-04-04:
> Latest testing implies this is not timing related. Running various approaches including bit bang on GPIO2 does not change behavior.
>
> However, suspicion is now on the data line voltage.
>
> Measuring [...]

Confirmed.  I was able to test it today and experienced the same loading / voltage drop.  Closing this as it's a hardware issue and unrelated to WLED. Sorry the goose chase!

closed 2023-04-04 by forkineye