wled/WLED #3108: Analog audio broken in 0.14b

5A52, 2023-02-26:
### What happened?

I have MAX9814 connected to esp32 GPIO36 and esp8266 A0. Both work wery well with any precompilled 0.13 binary with audioreactive mod. So hardware is ok, But none of precompilled 0.14 esp32 or even compilled from source with -D USERMOD_AUDIOREACTIVE works with analog microphone.

### To Reproduce Bug

Try to use analog mic 

### Expected Behavior

Mic to work

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b2 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-02-26:
Hi,

It could be that you are using a nightly build of SR WLED 0.13.3.x . The nightly build has some additional filters that sometimes allow to use broken analog hardware. Broken => instead of delivering a signal that is centered around 1.6V, some bad microphones are stuck at 3.3V or stuck at 0V in silence.

1) as there is no "precompilled 0.13 binary with audioreactive mod", are you referring to this one mabe? https://github.com/atuline/WLED/releases

2) Can you please add the build number of the SR-WLED that works for you?

Cheers

5A52, 2023-02-26:
Sorry for my bad english
Any 0.13 sounreactive from here https://wled-install.github.io works well on esp32 or esp8266 with max9814
Any 0.14b from there, or compilled by my self from this repository with soundreactive usermod does not work on esp32.  
It is possible i`m doing something wrong in pin settings, however nothing special, please see a screenshot
[image]

dosipod, 2023-02-27:
Max9814 ( also analog line in  ) works fine with us on esp32s with all   0.14b versions from  site https://wled-install.github.io/   . In some cases though we had to usb flash or recycle the power  as OTA  seems to mess with analog . For reduced issues go with digital mics

5A52, 2023-02-28:
Well, flashed just now WLED_0.14.0-b2_ESP32_ARE.bin directly from browser. Few restart\reconection, no reaction at all + checked different pins. Still does not work. Howeever 0.13, and other, non WLED firmare still works fine.  BTW checked on input pin with osciloscope - it is 0-3.3v amplitude with 1,55v silence level.

dosipod, 2023-03-02:
As of now I tested there ESP32 and they all worked even with MAX4466 which is even worst  bu
I assume you tested more then one esp32 and faced the same issue and if  that is the case then show link to your MCUs specs might be we have the same and list the exact version you are testing like   0.14.0-b2 (build 2302150) so we could try to replicate

5A52, 2023-03-03:
I have only one esp32 like this https://t.ly/T64C
With this fw https://wled-install.github.io/bin_dir/2_Sound_reactive_version/0.13.3/SR_ESP32_4MB_13.3_OB.bin and mic connected to GPIO36 soundreactive works.
With this fw: https://wled-install.github.io/bin_dir/1_Standart_version/Nightly_build/WLED_0.14.0-b2_ESP32_ARE.bin and same setup (settings on first picture) sound reactive does not work.
Also tryed MoonMood, MoonMood Nightly, and compilled by my self 0.14 betas with same result.  
[image]

dosipod, 2023-03-06:
It is a bit hard to confirm if you have only one MCU or mic  , see this page from SR wiki how to check for hw issues . One thing to test  is that the MIC is working without wled , there is a sketch to test that  
https://github.com/atuline/WLED/wiki/My--device--is-not-working

5A52, 2023-03-07:
Absolutely sure MIC (a got few) is working without wled. Ok let me explain a bit. I have a finished device - 16x16 matrix lamp, which works with non WLED but sound reactive firmware for a long time, and also works with any sound reactive wled 013 from https://wled-install.github.io. Works I mean just reacts on sound. But does not react on sound with any WLED 0.14 from same place or WLED 0.14 compilled by my self in vscode. So the only option is thet I mess something in sound settings (first picture) because thay differ a bit from WLED 0.13 settings.

dosipod, 2023-03-10:
We test a lot of mics and MCUs and our setups works on defaults ( we mostly use GPIO 36 for analog input  ) . We did face small issues when using OTA but the mics are sure to work after using usb to flash  . So we are also absolutely sure MICs works with 14 .
You could try again or get for reduced issues go with digital mics

closed 2023-03-28 by blazoncek

cdalexndr, 2024-01-13:
I've had the same issue in **0.14.0**, but works in **0.14.1-b3**
ESP32 DEVKIT V1