wled/WLED #3112: Gamma correction does not apply to UDP realtime

CarbonNeuron, 2023-03-02:
### What happened?

When using DRGB, or WARLS (It may be on all of the UDP realtime packet types, I haven't tried all of them), Gamma color correction is not applied to the output even when the respected options are turned on (Use Gamma correction for color).

### To Reproduce Bug

Sending #FFA500 to the LEDs over UDP realtime using DRGB or WARLS produces a non color corrected yellow color.
Shown Here:
[image]

### Expected Behavior

I expect that gamma correction is applied to UDP realtime LEDs when the option is enabled. Here is a picture of what it looks like when applying gamma correction on the sender side.
[image]
 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED v0.13.3 (Build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

CarbonNeuron, 2023-03-02:
I've found the option that disables realtime gamma correction under Settings -> Sync Interfaces -> E1.31 Info. This is handled correctly on WLED v0.13.3 (Build 2208222), but not on the latest development build. In the GetV() function an exception is thrown that says d.Sf.PY.value = 0 cannot set value on undefined. PY does not exist, so it can't load any settings past that.

closed 2023-03-02 by CarbonNeuron

blazoncek, 2023-03-02:
Clear cache ans local storage.
Settings save works as expected.