wled/WLED #3113: multiple pins; fire2012 leads to incorrect pixel used for effect on first output

170bgrd, 2023-03-05:
### What happened?

ESP32 from Amazon
using 2 strips of WS2815; 110 LEDs on each
strip 1; pin 12
strip 2; pin 26
using a 12v 3a power supply; 
have a relay controlling power pin 27

When using fire 2012 (my intended effect) then on strip 1; the last few pixels are illuminated with a flickering effect
if I reverse the effect on the strip then the random pixels also move to the other end of the strip. If I leave it reversed and tell wled to skip the first few (3 or 4) LEDs then the randomly flickering leds also moves by this small number of pixels

### To Reproduce Bug

use fire2012 effect with two gpio setup.

### Expected Behavior

the effect should peter-out and the end of the strip should mostly be dark - unless a very intense "flame" illuminates it

### Install Method

Binary from WLED.me

### What version of WLED?

14.0 b1; but have also tested 0.12 with the same result

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

170bgrd, 2023-03-05:
I've tried a couple of other pins; the artefact (group of flickering incorrect pixels) always follows the first segment whichever GPIO pin is assigned.

blazoncek, 2023-03-05:
Check your segment set up.

170bgrd, 2023-03-05:
Hi Blazencek

I have; here is my config. I have only changed network names

R

{"rev":[1,0],"vid":2212222,"id":{"mdns":"WLED","name":"WLED","inv":"Light"},"nw":{"ins":[{"ssid":"SSID","pskl":16,"ip":[0,0,0,0],"gw":[0,0,0,0],"sn":[255,255,255,0]}]},"ap":{"ssid":"WLED-AP","pskl":8,"chan":1,"hide":0,"behav":0,"ip":[4,3,2,1]},"wifi":{"sleep":false},"hw":{"led":{"total":220,"maxpwr":2500,"ledma":255,"cct":false,"cr":false,"cb":0,"fps":42,"rgbwm":3,"ld":false,"ins":[{"start":0,"len":110,"pin":[26],"order":0,"rev":true,"skip":0,"type":22,"ref":false,"rgbwm":0},{"start":110,"len":110,"pin":[25],"order":0,"rev":true,"skip":0,"type":22,"ref":false,"rgbwm":0}]},"com":[],"btn":{"max":4,"pull":true,"ins":[{"type":2,"pin":[0],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]}],"tt":32,"mqtt":false},"ir":{"pin":-1,"type":0,"sel":true},"relay":{"pin":27,"rev":true},"baud":1152,"if":{"i2c-pin":[-1,-1],"spi-pin":[-1,-1,-1]}},"light":{"scale-bri":100,"pal-mode":0,"aseg":true,"gc":{"bri":1,"col":2.8},"tr":{"mode":true,"dur":7,"pal":0},"nl":{"mode":1,"dur":60,"tbri":0,"macro":0}},"def":{"ps":1,"on":true,"bri":128},"if":{"sync":{"port0":21324,"port1":65506,"recv":{"bri":true,"col":true,"fx":true,"grp":1,"seg":false,"sb":false},"send":{"dir":false,"btn":false,"va":false,"hue":true,"macro":false,"grp":1,"ret":0}},"nodes":{"list":true,"bcast":true},"live":{"en":true,"mso":false,"port":5568,"mc":false,"dmx":{"uni":1,"seqskip":false,"addr":1,"mode":4},"timeout":25,"maxbri":false,"no-gc":true,"offset":0},"va":{"alexa":false,"macros":[0,0],"p":0},"mqtt":{"en":false,"broker":"","port":1883,"user":"","pskl":0,"cid":"WLED-fa5238","topics":{"device":"wled/fa5238","group":"wled/all"}},"hue":{"en":false,"id":1,"iv":25,"recv":{"on":true,"bri":true,"col":true},"ip":[192,168,178,0]},"ntp":{"en":false,"host":"0.wled.pool.ntp.org","tz":0,"offset":0,"ampm":false,"ln":0,"lt":0}},"ol":{"clock":0,"cntdwn":false,"min":0,"max":29,"o12pix":0,"o5m":false,"osec":false},"timers":{"cntdwn":{"goal":[20,1,1,0,0,0],"macro":0},"ins":[]},"ota":{"lock":false,"lock-wifi":false,"pskl":7,"aota":true},"um":{}}

blazoncek, 2023-05-27:
Please post your preset. The config is ok.
You may also want to update to most recent build (from @srg74 repository).

closed 2023-06-12 by 170bgrd