wled/WLED #3119: TM1814 losing data line

cthunberg, 2023-03-09:
### What happened?

I have 2 dig quads setups with 24v TM1814. One is a bench setup to mitigate concerns with voltage drop. They both appear to revert to "demo" mode periodically and flash red when on and off (in WLED). The bench setup is running current code, as of March 8, with DEBUG_WLED. This issue manifest itself on 0.13.3 and 0.14.0-b1 and b2/current. I have not tried to downgrade to various other versions but more than happy to test with some guidance. 

### To Reproduce Bug

This bug is not reproducible on demand but 2 24v dig quad/esp32 implementations exhibit similar behavior. 

### Expected Behavior

Does not revert to demo mode. 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.3, 0.14.0-b1, 0.14.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Issue manifests near the end of this log

JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Local time: 05:42
---DEBUG INFO---
Runtime: 18398070
Unix time: 1678340526,103
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1783
UM time[ms]: 0/1
Strip time[ms]: 0/2
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
---DEBUG INFO---
Runtime: 18428083
Unix time: 1678340556,116
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1776
UM time[ms]: 0/1
Strip time[ms]: 0/1
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Local time: 05:43
---DEBUG INFO---
Runtime: 18458096
Unix time: 1678340586,129
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1776
UM time[ms]: 0/1
Strip time[ms]: 0/22
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
---DEBUG INFO---
Runtime: 18488109
Unix time: 1678340616,142
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1782
UM time[ms]: 0/1
Strip time[ms]: 0/7
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Local time: 05:44
---DEBUG INFO---
Runtime: 18518122
Unix time: 1678340646,155
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1776
UM time[ms]: 0/1
Strip time[ms]: 0/10
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
---DEBUG INFO---
Runtime: 18548135
Unix time: 1678340676,168
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1776
UM time[ms]: 0/1
Strip time[ms]: 0/3
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Local time: 05:45
---DEBUG INFO---
Runtime: 18578148
Unix time: 1678340706,181
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1781
UM time[ms]: 0/1
Strip time[ms]: 0/1
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
---DEBUG INFO---
Runtime: 18608161
Unix time: 1678340736,194
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1782
UM time[ms]: 0/1
Strip time[ms]: 0/45
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer locked. (17)
JSON buffer size: 2088 for request: 3
JSON buffer released. (17)
Local time: 05:46
---DEBUG INFO---
Runtime: 18638174
Unix time: 1678340766,207
Free heap: 201240
Wifi state: 3
State time: 0
NTP last sync: 4398
Client IP: 192.168.100.158
Loops/sec: 1783
UM time[ms]: 0/1
Strip time[ms]: 0/2
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

cthunberg, 2023-03-09:
LEDs are Teopek IP20, https://www.aliexpress.us/item/3256804115281569.html

Chris87-ux, 2023-03-09:
I supose, the same appear to my SK6812 RGBW 12V strip. Randamly it only allowes me to control only 20 of my 142 LEDs on this strip and want take my changes from the settings, unless I do a complete restart of the controller. 
I Use a WT32-ETH with 0.14.0-b1.

cthunberg, 2023-03-09:
When the strip flashes red, the first led (or 6 in the case of 24v TM1814) are skipped. This behavior is mirrored when changes are made to the 'Sync setup' option in WLED. Video: https://photos.app.goo.gl/5M7pWDiwgDh4BpMh7

cthunberg, 2023-03-10:
Turning off/unchecking all of the Sync options (e.g. sync groups, instance list, and realtime) as well as unchecking the Enabled time macros seems to have had a positive impact on the problem. The TM1814 LEDs default to being On and in demo mode when they don't have a data line unlike the SK6812s and maybe others. This may mean that a bus reset or reloading of settings periodically leads to a lost data line the uniquely affects the TM1814s? Would random traffic to sync ports cause WLED to think it needs to change settings? Or enabled but unconfigured sunset or sunrise macros cause something similar?

msfuture, 2023-04-29:
In my case the cause for the red blinks seems to be an integration in Home Assistant. As soon as I add my WLED TM1814 strips to Home Assistant the red blinking begins. We already discussed this an HA side here:
https://github.com/home-assistant/core/issues/90676

It seems that HA is polling a huge amount of data, so the ESP is busy answering these requests and the strip is losing connection meanwhile. Same happens when saving the configuration using the WLED webinterface. 
Is there any chance to give the data output for the stripts more prio?

blazoncek, 2023-05-27:
@cthunberg please connect oscilloscope to data line and monitor it.
Make sure you select correct LED type in led settings and check that "Off refresh" is checked for TM1814.

xeetsh, 2023-08-11:
@cthunberg do you have any way to directly reproduce the issue? I have the same issue but it happens like one time in 5 hours of operation. I have an oscilloscope and could have a look at the data line but this will be hard without having any means of reproducing this.

sdwilsh, 2023-08-19:
I can get this to happen if I just go to LED Preferences and then press save about 25% of the time on 0.14.0-b4.  I don't have an oscilloscope, but I've got a friend who might be able to help me with that in a week.

rgregrowe, 2023-09-08:
I've been chasing this problem for the last couple weeks thinking the flashing red was due to cable lengths. I'm trying to use  TM1814 pucks in landscape lights. I've tried putting in QuinLED Data Boosters in several locations with power injection and the problem persists. I get a flicker on the first puck in the line and the rest of the pucks flash red. This happens every couple minutes. I found the HA issue report and sure enough once I removed the WLED integration and restarted HA, I've had no problems with the red flash. I've started a MQTT integration for now and that's working fine. I'm using a Planet Laterna controller which is also running some analog RGB lights which don't seem to have any issues. Latest WLED and HA software.

blazoncek, 2023-09-18:
No response received, closing.

closed 2023-09-18 by blazoncek