wled/WLED #3122: Version 0.13.3 0.14.0-b1 not compileable

nauen, 2023-03-10:
### What happened?

Hey guys,

I'm very happy about this project. But your last versions are not compileable out of the box.
Once the error message about the forked async library.
Second NeoTm1814Settings is not available "there are no arguments to 'NeoTm1814Settings' that depend on a template parameter", so I tried to comment out but then
Third 'DotStarSpi5MhzMethod' was not declared in this scope

I'm stopping here to figure why this version is not running. I just wanted to try it out of the box to check a dev board and not be able since hours.

frustrating...

### To Reproduce Bug

just compile

### Expected Behavior

compile is working

### Install Method

Self-Compiled

### What version of WLED?

0.14.0-b1, 0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-03-10:
Hi, thanks for your report.
Actually I am wondering a bit how you are building from the source code. Could it be you try using Arduino IDE, instead of [VSCode+platformio](https://kno.wled.ge/advanced/compiling-wled/#compilation-guide-platformio)? Arduino IDE is not recommended any more. 

Please give us some more details, like the build log after you hit "PlatformIO:Build".

blazoncek, 2023-03-11:
You build environment is broken. Please update it and compile again.
Samples are included.

As @softhack007 said, if you are using Arduino IDE you will need to research for yourself how to update it or switch to (MS VSC +) PlatformIO.

nauen, 2023-03-11:
Alright, sorry I missed the move to platform.io.

It's a bummer for me, one of my favorite feature of wled was, that its all-in-one and still able to extend in Arduino IDE.
Platform.io add so much more complexity to the project. I understand that its helpful to work on the features. For my case it's just one tool of many and now I have to add another one.

So I will use an older versions until I have time to figure platform.io or changing the code.

So it's not a bug. It's a feature.

Thank you for keeping wled up! I'm working with it since many years and with many different functions.

softhack007, 2023-03-11:
Closing, as it's not a bug.

closed 2023-03-11 by softhack007

blazoncek, 2023-03-11:
> Platform.io add so much more complexity to the project.

Does it? I switched from Arduino IDE two years ago and never looked back. So much easier and feature complete.