wled/WLED #3123: Unable to control CCT on SMD2835

kub3let, 2023-03-12:
### What happened?

I bought this ceiling lamp from banggood and replaced the WB3L for a ESP8266-12S: BlitzWill BW-CLT1

I set it up to have 2 segments, one for the RGB SMD5050 which works just fine including effects and one for the CCT SMD2835.

Now the problem for me is however I configure WLED, it's always that both sliders on the left (not the upper right brightness slider) only control the brightness of the lamp, but it does not change the white balance e.g. cold to warm white.

Is this a know bug, did I configure something wrong ? 

### To Reproduce Bug

LED output 1: PWM RGB GPIOS: 15, 14, 12
LED output 2: PWM CCT GPIOS: 4, 13
Make a segment for each output: check

### Expected Behavior

Balance slider should change the lamp from cold white to warm white

### Install Method

Binary from WLED.me

### What version of WLED?

v0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Purchase link: https://www.banggood.com/BlitzWill-BW-CLT1-LED-Smart-Ceiling-Light-30cm-with-Main-Light-and-RGB-Atmosphere-Light-2700-6500K-Adjustable-Temperature-APP-Remote-Control-Optional-and-DIY-Scene-Mode-p-1793809.html

Tasmota reference: https://templates.blakadder.com/blitzwolf_BW-CLT1.html

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-03-12:
Are you sure this is on the compatible list?
CCT is working as expected on compatible hardware.

kub3let, 2023-03-12:
> Are you sure this is on the compatible list?

These are general analog led strips that are controlled via mosfet, which is on the supported list (analog pwm).

They are not addressable so most effects won't work, but CCT should still work as is.

blazoncek, 2023-03-12:
Then your wiring or set-up is incorrect. 🤷‍♂️

kub3let, 2023-03-16:
I currently have no time to debug this, once I have I will flash Tasmota to see if it makes any difference.

closed 2023-03-16 by kub3let

p1otr, 2023-10-28:
I have the same lamp (BW-CLT1) with the same module (ESP-12S) and I can confirm what kub3let reported.
Is there any way to get CCT working (it works fine on Tasmota).