wled/WLED #3124: "Transitions" in WLED for everyday home usage (Arhitecture / Household)

JKSniper1, 2023-03-12:
WLED as is, has tons of possibilies regaring very colorful blinking efects, but for everyday usage, for example for outdoor house lights, indoor stairs, bed, bathroom mirror etc. this are too colorful effects and are not an option, except if you're a teenage gamer or there is a party or some special holiday mood. On other days, you just need white or single color light that you turn ON/OFF every now and then.  

What would be really nice to have is TRANSITION effects from OFF->ON and from ON->OFF. 
- Right now Transition is already an option in SETTINGS -> TRANSITIONS -> CROSSFADE. But this option is kinda "basic" for digital led strips and it's also not so flexible to use. 

What would be a nice to have is another section, just same as COLORS and EFFECTS with name TRANSITIONS, where you could select TYPE of transition and its setting. It should work with any effect, color etc. that already is there, just to enable transition between OFF -> ON in a certain way. 

Transitions that would be great to have as starter: 
- Crossfade (we already got that) 
- "Train" of light from one side of segment to another (LED) 
- "Reverse train" 
- "Center train" Turns on the middle first and then LEDs activate in both sides from the center
- NEON light imitation (segment divided in smaller sections, which would turn on in erratic random manner, like turning on old garage neon lights)
- Pixalate (Turn on random pixels on random time in some limited interval) 
- etc. 

And reverse transitions for ON -> OFF. That way, even having just single color, would be super interesting, as there would be "animation" for turning on. 

This is very hard or impossible to do with playlists, as effects are really meant as something long lasting, while here we are only looking for a transition from off-on  and on-off that would last only from fraction of a second up to a few seconds. 

I'm adding a video, where in effects you would select a single color, color would be something yellow and for transition a "train". 
https://user-images.githubusercontent.com/8069093/224567408-6dd413ca-f635-4128-b1ad-f90249982a24.mp4

blazoncek, 2023-03-12:
That is on the roadmap for very long time now and there is no need to open an issue for it.
The problem is it is not trivial to do effect transitions.

eokgnah, 2023-04-28:
Hi JKSniper1,

i searched for something like this as well. And i think i found it already in WLED!
Take a look at the "Percent" Effekt.
Make segments.
0 is off. 
1-99 is fill the Segment 1% to 99%.
100 is the fully lit Segment.
101-199 is switching off the lit Segment 0% to 100%.
200 is all off like 0.
Maybe play with Segment Mirroring and Reversing to get the CenterTrain Effects you wish.
Trigger with mqtt or Web-Api and you are set. ;-)

JKSniper1, 2023-04-28:
Hey! Thanks for that info. 

I've tested right now and this does not work with neither STAIRCASE usermod or normal "switch" on PIR sensor. WLED won't do playlist but just a single effect when triggered.

- If you choose this Percent effect, it won't be dynamic, it will just lit preset value (meaning from off leds to full leds, all or partial strip at once). You can't dynamically change it, because for it to work, it has to go from 0 to 100 to do transition effect and it also doesn't take any speed value (which would be nice, as now it's really fast (when it works, by manually selecting the effect via UI)). 
If I would program this outside WLED somehow as series of commands and trigger animation via mqtt or Web-Api, it's not reliable and takes too much time. I need reaction the very same moment the PIR sensor is triggered, so what I would basically need is have this inbuilt in WLED code. 

While I love WLED and I'm amazed how it works, how it's wrapped up, how the UI is made, settings and all the features it does and appreciate all the hard work people have put into it, I can't use it for any "serious stuff" as is, as I can't get from  OFF -> dyamic effect -> ON (just to some fixed color) -> dynamic effect -> OFF. It's suitable for bliking party colorful lights, usable for... I don't know. maybe gaming PC enclosure? 

I need to check if I can maybe do my own "effect" that would have this transition hardcoded in it. Or doing custom usermod, to just play an animation I hardcode in somehow.

blazoncek, 2023-04-29:
@JKSniper1 what you are describing are not *transitions*. Those are "chains of effects".

You can achieve the desired outcome by carefully timing the effects, create presets with such timing and add them (presets) to playlists. I agree that not every situation can be accomplished in that way but most do.

The scenario from video posted can be easily achieved by playlist using Solid and Wipe effects. Using several segments.

SpeedyRS2, 2023-05-10:
> The scenario from video posted can be easily achieved by playlist using Solid and Wipe effects. Using several segments.
This is the way I tried it. I have 3 segments at a wall, but how can I achieve it so that each of the 3 segments starts slightly delayed, while the segment before is not yet finished?

blazoncek, 2023-05-10:
Use different *timebase* for each preset. You may also need to create multiple presets each containing configuration for each segment.
Check [KB](https://kno.wled.ge/interfaces/json-api/#state-object).

As noted above not all variations may be achievable that way.

blazoncek, 2023-05-27:
Closing as playlists are used for time sensitive "chains of effects".

For a particular "effect" that cannot be produced by playlist(s), please open a new issue with detailed description and possible video.

closed 2023-05-27 by blazoncek