wled/WLED #3125: Brightness min/max limits on each segment

Boolkinator, 2023-03-13:
Please add the ability to set minimum and maximum brightness limits for each segment. This can be useful when using a wall switch/dimmer. I'm currently having a problem with the LEDs completely off at "0" brightness while the switch stays on. This is confusing. You press on, off, and nothing happens. If there was a brightness limiter, then at the minimum value the LED could remain visible, and the control using the wall switch remained intuitive.
Another concept can be applied: At minimum brightness (eg 1%), the power state always changes to the off. When turned on, the brightness is applied as before turning it off, but not less than the specified threshold.

Boolkinator, 2023-03-13:
In this case, I use KNX switches. Integration through Home Assistant.

blazoncek, 2023-03-14:
Thank you for the recommendation, unfortunately this is very difficult to implement.
Recommendation I can give is to use potentiometer in such case.
IMO this will not be implemented at any point.

blazoncek, 2023-11-16:
The closest is #2364 done in 8f7f9ec3676af6d1da8a9b5114a43d53dbcc0fd7

closed 2023-11-16 by blazoncek