wled/WLED #3126: usermod_mpu6050_imu.h won't compile

clayclai, 2023-03-14:
### What happened?

usermod_mpu6050_imu.h won't compile as it exists.
Line 212 
JsonArray imu_meas = user.createNestedObject("IMU");
needs to be changed to:
JsonObject imu_meas = user.createNestedObject("IMU");

With that change, I can build wled with usermod_mpu6050 and see the results in /json/info
Where I can really use your help is here: How do I make the 6 outputs of the mpu6050 available in Custom Effect?
Is there a standard, or wled best practice way to do that? I've been trying to trace how /json/info/led/count makes its way to ledCount used in Custom Effect but can't quite make it out. Also how is wled.jsaon generated? Where does it come from? 

### To Reproduce Bug

try to build wled with usermod_mpu6050, as per instructions in Readme.md with any version before WLEDMM, which also has fixed this problem.

### Expected Behavior

expect to be able to build wled with usremod_mpu6050 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32, ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-03-14:
Please contact usermod creator.
BTW version 0.13.3 is no longer maintained, please use latest source.

softhack007, 2023-03-14:
Usermods are contributed by individual user, and not maintained by the WLED core team. Please contact the usermod author to get help with your problem.

For integration in your own effect, take a look at the "games" usermod that you find here:
https://github.com/MoonModules/WLED/blob/mdev/usermods/usermod_v2_games/usermod_v2_games.h

You dont need to fiddle around with json arrays to get IMU data, as there are direct way s for communication between usermods, like `usermods.lookup`. For example

```C++
void setup() { 
#ifdef USERMOD_MPU6050_IMU 
     IMU = (MPU6050Driver *)usermods.lookup(USERMOD_ID_IMU); 
   strip.addEffect(255, &mode_IMUTest, _data_FX_MODE_IMUTest);
#endif
```

```C++
uint16_t mode_3DIMUCube(void) { 
float yaw = 0; float pitch = 0; float roll = 0; 
#ifdef USERMOD_MPU6050_IMU 
if (IMU != nullptr) { 
   yaw = -IMU->ypr[0]; 
   pitch = IMU->ypr[1]; 
   roll = IMU->ypr[2]; 
} 
#else 
   //simulate rotation 
    yaw = (fmod(SEGENV.call, 360)-180) / (180/M_PI); //-180 .. 180 
    pitch = yaw; roll = yaw; 
#endif
```

clayclai, 2023-03-16:
Thank you for your attention and advice on this, as well as your good work
on wled.

Clay

Clay Claiborne, Director
Vietnam: American Holocaust <http://VietnamAmericanHolocaust.com>
Linux Beach Productions
Venice, CA 90291
(310) 581-1536

Read my blogs at the Linux Beach <http://claysbeach.blogspot.com/>
<http://wlcentral.org/user/2965/track>

blazoncek, 2023-03-19:
Fixed by #3138

closed 2023-03-19 by blazoncek