wled/WLED #3128: [14.0-b2] WLED stop working almost immediately on ESP32 when RX pin (GPIO03) is not pulled down (and USB IC is not attached)

wled-install, 2023-03-14:
### What happened?

On a ESP32 based board without USB IC WLED software stops working almost immediately when RX pin (GPIO03) is not pulled down (and USB IC is not attached). After power-on WLED starts corerctly, it can be seen tat LEDs are controlled for a short time period, then it stops (no LED control signal, no GUI can be reached).
The issue is with current 0.14-b2 build. No issue with 13.3 Release.

### To Reproduce Bug

1) Take ESP32 based board without USB IC an nothing connected to RX (GPIO3) Pin.
2) Take and connect USB adapter
3) Erse flash and flash 14.0-b2 build
4) Disconnect USB adapter
3) Attach some leds to default GPIO to see if they are controlled
4) Disconnect from the power board and connect again
5) You will see tat LEDs are controlled for a short time, but no WLED-AP etc. since ESP32 stop working
6) Tie GPIO3 (RX Pin) to Ground or attach USB adapter and repeat power off7on again. Now is everything fine.

### Expected Behavior

Full functionality independent on what is attached or not to RX Pin

### Install Method

Self-Compiled

### What version of WLED?

0.14-b2 (dev branch taken on 14. of march 2023)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wled-install, 2023-03-15:
could be HW +SW combination, closed till clarified in details

blazoncek, 2023-03-16:
This may not be WLED SW related as there were no change in Serial processing in 0.14.
It may have to do with libraries used but that is beyond the scope of WLED.
My recommendation would be, if you design PCB with ESP and no USB-to-serial chip, pull-down GPIO3 (also pull-up GPIO0, GPIO1 and, in some instances, GPIO2).

softhack007, 2023-03-16:
* last change to wled_serial.cpp (between 0.13 and 0.14) was 
https://github.com/Aircoookie/WLED/pull/2667

* this line seems a bit dangerous, as it runs infinitely if there is continuing input on RX (like noise) 
https://github.com/Aircoookie/WLED/blob/d1fed11d0d0e93b727d8156e9523f3d1a47ce05f/wled00/wled_serial.cpp#L82

* this line explains why the problem does not occur when gpio03 is configured as button https://github.com/Aircoookie/WLED/blob/d1fed11d0d0e93b727d8156e9523f3d1a47ce05f/wled00/wled_serial.cpp#L70-L73

softhack007, 2023-03-16:
A few ideas - but not sure if that would break other functionality
* add `if (!Serial) return;` to catch situations where serial was not initialized (like CDC USB without cable connection)
* add a limit to the `while(Serial.availeable())` loop, for example 10 milliseconds

To be discussed...

softhack007, 2023-03-16:
Possibly not a SW bug, but let's discuss if we can avoid a lockup somehow

wled-install, 2023-03-16:
pinMode(hardwareRX, INPUT_PULLDOWN); delay(1) or external pull-down or pull-up helps at low noise energy. It must be sufficient for household’s environments. For more noisy environments either dedicated EMC measures in HW needed or the possibility to switch off serial functions if not used.

So my proposal would be at least to implement pinMode(hardwareRX, INPUT_PULLDOWN); delay(1) in SW. Should not have any negative impact, easy to implement and must be enough for mainstream. Maybe additinal setting in GUI to switch serial off.

blazoncek, 2023-03-16:
Please add DEBUG_PRINT statements around the code @softhack007 mentioned and use net debug functionality to capture debug output.

blazoncek, 2023-03-16:
It appears that random (or EMI) data triggers Serial input.

softhack007, 2023-03-19:
> Maybe additional setting in GUI to switch serial off.

Its not by GUI yet, but you could un-define WLED_ENABLE_ADALIGHT in wled.h
This disables reading from hardwareRX in `handleSerial()`.

https://github.com/Aircoookie/WLED/blob/cac51737cb5f45091446cc9c101b0770a77520df/wled00/wled.h#L33

<br/>

With commit fd8920923346a5131d3ebd212090a48731d856a2, I've added the flag WLED_DISABLE_ADALIGHT so that you can disable use of RX pin in your custom platformio_override.ini build env.

wled-install, 2023-03-19:
> With commit [fd89209](https://github.com/Aircoookie/WLED/commit/fd8920923346a5131d3ebd212090a48731d856a2), I've added the flag `WLED_DISABLE_ADALIGHT` so that you can disable use of RX pin in your custom platformio_override.ini build env.

Yeah, I've seen this #define in the code, thank you!

wled-install, 2023-03-19:
> It appears that random (or EMI) data triggers Serial input.

Yes it is definately. For low noise energy the solution from softhack is sufficient. For more noisy enviroment we can turn serial off by define (or maybe in future someone make a switch in the GUI). I think the issue can be closed.

closed 2023-03-19 by wled-install

softhack007, 2023-03-19:
> 

Thanks for the feedback :-) There is only one downside of `-D WLED_DISABLE_ADALIGHT` - it also disables IMPROV, which works over serial. So web installers cannot configure Wifi settings after install any more when using this option.