wled/WLED #3131: 2D Matrix over 1024px not working correctly

gwwiktor, 2023-03-15:
### What happened?

Bottom half of the display does the same thing that the last correctly working LED and animations are doubled on the left and right side when 2D matrix contains more than 1024 pixels. My example contains 1800 pixels configured in 60x30 2D Matrix.

### To Reproduce Bug

1. Install WLED v0.14.0-b2 (build 2303050) or v0.14.0-b1
2. Configure outputs for 1800LEDs
    [image]
3. Configure 2D Matrix to 60x30 pixels, Vertical, Serpentine checked
    [image]
4. Try to use Effects and observe

### Expected Behavior

2D Matrix is expected to display chosen animation on the whole surface and each pixel is expected to be driven independently

### Install Method

Binary from WLED.me

### What version of WLED?

WLED v0.14.0-b2 (build 2303050)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Similar issue was mentioned before, but in fact it is not fixed in the newest build
[https://github.com/Aircoookie/WLED/issues/3120](url)

Bottom half working incorrectly and the arrow pointed to the last working LED:
[image]

[image]

Doubled animation on the upper half of the matrix and bottom not working:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-03-15:
Please send pictures of actual LED output, not peek preview.

gwwiktor, 2023-03-15:
Thank you for such fast response and also forgive me my lack of consideration. I did not think of possibility of broken peek preview and I didn’t try connecting LEDs as I thought preview reflects actual output data. I’ll try to do better next time

closed 2023-03-15 by gwwiktor

blazoncek, 2023-03-16:
@ewoudwijma could you look into this?

MoonModules, 2023-03-16:
Change the following in ws.cpp / sendLiveLedsWs :

  const uint16_t MAX_LIVE_LEDS_WS = strip.isMatrix ? 4096 : 256; //WLEDMM use 4096 as max matrix size

instead of 1024

blazoncek, 2023-03-16:
> Change the following in ws.cpp / sendLiveLedsWs :
> const uint16_t MAX_LIVE_LEDS_WS = strip.isMatrix ? 4096 : 256; //WLEDMM use 4096 as max matrix size

Unfortunately this may have adverse consequences as the size of 4096 may still not be adequate. It will also needlessly send *all* matrix LEDs to a peek client possibly overwhelming MCU.
I'd much rather see a scaled approach in the same way current 1D LEDs are handled.

MoonModules, 2023-03-17:
Maybe ... but as long as that functionality is not in place, 4096 is better then 1024, note that only total leds are sent so if you have a matrix of 16x16 only 256 pixels are sent, so it's in your own hands ;-)
4096 working fine in WLED MM until now.

blazoncek, 2023-03-17:
I've now written matrix scaling if matrix is bigger than 256 pixels.
It works for any size of matrix up to 8192 pixels.

It does not look best when matrix is above 256 pixels, but reduces strain on ESP so ESP8266 can also provide peek on matrices of 512 pixels or more.

It will be available after a while.

MoonModules, 2023-03-17:
Nice!

Less nice is it does not look best ...

Can't you make it conditional on esp8266? And make it higher then 256 bytes, 1024 sounds better

kmatokov, 2023-07-31:
HI, i use 4x1052 out in matrix, it look correct on led matrix but visualization is wrong, not showing configured led but some reduced and shifted. when use low amount of pixel it work correct. not so important but when work remote it would be more handy to see what is currently playing.