wled/WLED #3132: DMX_MODE_PRESET lags out the ESP32

strikeout, 2023-03-15:
### What happened?

Im using an external hardware DMX Controller to send ArtNet (or e1.31) packets to WLED to select various presets.
The hardware controller sends its values aprox 2 times per second. This  completely overwhelms the ESP32 - especially when the UI is open. You can see the same preset being selected again and again, lagging out the controller - until the WS connection drops completly.

### To Reproduce Bug

Use DMX_MODE_PRESET, send repeated message with the same channel/value, see the system lag to death.

### Expected Behavior

I expect DMX messages with the same channel/value pair to be dropped and not be re-applied.

### Install Method

Binary from WLED.me

### What version of WLED?

v14.0.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Would a tiny buffer to save the last DMX message for comparison to current received message help? 
Then do a simple `if (lastDmx == currentDmx) return;`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-03-15:
Unfortunately there is no logic (or state history) within WLED to support such behaviour.
Please update settings on your DMX equipment as MCU is already on the verge of capacity handling all possible uses.

If, on the other hand, you know how to solve the issue then please make a PR.

blazoncek, 2023-03-16:
Check JSON API state & info responses. "ps" and "pl".
One is for playlist the other for preset.

strikeout, 2023-03-16:
Is my assumption that the global `currentPreset` is populated and usable in the following form correct?

```
case DMX_MODE_PRESET:       // 2 channel: [Dimmer,Preset]
      ...
      if (currentPreset != e131_data[dataOffset+1]) {
        applyPreset(e131_data[dataOffset+1], CALL_MODE_NOTIFICATION);
      }
      ...
```

mxklb, 2023-03-23:
Thanks for testing. Preset mode is new and wasn't tested that much :smile: 

``` 
    case DMX_MODE_PRESET:       // 2 channel: [Dimmer,Preset]
      if (uni != e131Universe || availDMXLen < 2) return;
      if (currentPreset() != e131_data[dataOffset+1]) {
        applyPreset(e131_data[dataOffset+1], CALL_MODE_NOTIFICATION);
      }
      if (bri != e131_data[dataOffset]) {
        bri = e131_data[dataOffset];
        strip.setBrightness(bri, true);
      }
      return;
      break;
```
Like with brightness, with `currentPreset()` this shall do the trick. @blazoncek do you know a smart way to get that `currentPreset()` at this position in e131.cpp? Are there any getter methods? If I find time to I'll have a look .. the issue should be quite easy to fix this way.

mxklb, 2023-03-23:
@strikeout sorry didn't looked into your pr, looking good!

blazoncek, 2023-03-23:
`currentPreset` is a global variable. No getter or setter method needed.

closed 2023-05-24 by strikeout