wled/WLED #3139: Applying changes to formatted presets.json makes presets stop working

cr1tbit, 2023-03-19:
### What happened?

In my project, after creating a preset and WLED config, I exported them and put in my /data folder in platformio, so they become the default setting. In order to increase readability, I applied "auto format" option for those json files.

After that, all attempts at editing presets result int "Sorry there was an issue loading your presets!" error.

After looking into presets.json via <WLED_IP>/edit, it was indeed damaged (attached at the bottom of the issue)

Editing config works fine even if it was formatted the same way.

### To Reproduce Bug

Insert a formatted "presets.json" into /data folder on platformio, build and flash the filesystem image

As an example you can use my preset.json:

```{
    "0": {},
    "1": {
        "on": true,
        "bri": 110,
        "transition": 7,
        "mainseg": 0,
        "seg": [
            {
                "id": 0,
                "start": 0,
                "stop": 256,
                "grp": 1,
                "spc": 0,
                "of": 0,
                "on": true,
                "frz": false,
                "bri": 255,
                "cct": 127,
                "col": [
                    [
                        255,
                        20,
                        255
                    ],
                    [
                        0,
                        0,
                        0
                    ],
                    [
                        0,
                        0,
                        0
                    ]
                ],
                "fx": 115,
                "sx": 128,
                "ix": 128,
                "pal": 1,
                "c1": 128,
                "c2": 128,
                "c3": 16,
                "sel": true,
                "rev": false,
                "mi": false,
                "o1": false,
                "o2": false,
                "o3": false,
                "si": 0,
                "m12": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            }
        ],
        "n": "Blends"
    }
}```

### Expected Behavior

The firmare should either work fine with formatted json, or remove all whitespaces before applying changes, so that we're not completely locked out from presets feature.

Honestly speaking, it'd be even better to completely reject formatted json, as in our case all the devices were already programmed and closed when the issue was detected.

Another workaround would be to allow purging the presets without resetting cfg.json.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-1b

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

How the presets.json looks after attempting a change from firmware (this is obviously invalid JSON)

```
{
    "0": {},
         {
        "on": true,
        "bri": 110,
        "transition": 7,
        "mainseg": 0,
        "seg": [
            {
                "id": 0,
                "start": 0,
                "stop": 256,
                "grp": 1,
                "spc": 0,
                "of": 0,
                "on": true,
                "frz": false,
                "bri": 255,
                "cct": 127,
                "col": [
                    [
                        255,
                        20,
                        255
                    ],
                    [
                        0,
                        0,
                        0
                    ],
                    [
                        0,
                        0,
                        0
                    ]
                ],
                "fx": 115,
                "sx": 128,
                "ix": 128,
                "pal": 1,
                "c1": 128,
                "c2": 128,
                "c3": 16,
                "sel": true,
                "rev": false,
                "mi": false,
                "o1": false,
                "o2": false,
                "o3": false,
                "si": 0,
                "m12": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            },
            {
                "stop": 0
            }
        ],
        "n": "Blends"
    }
,"1":{"on":true,"bri":110,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":256,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,20,255],[0,0,0],[0,0,0]],"fx":115,"sx":128,"ix":128,"pal":1,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Blends"}}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-03-19:
Do not use /data folder for making file system image.
It contains files that will (when uploaded) interefere with built-in files created during compilation.

Also, as suggested in [KB](https://kno.wled.ge), never prettify JSON for use with WLED.

Whenever you have *corrupt* presets.json file, just delete it using /edit page and reupload good one. This can be done in AP mode using backup/restore functionality.

blazoncek, 2023-04-24:
Closing as working with prettified JSON is not intended.

closed 2023-04-24 by blazoncek