wled/WLED #3140: Support for LilyGO T-ETH-POE board.

cheeseshark, 2023-03-19:
### What happened?

Do you have support for the LiliGO T-ETH-POE board?

I was able to upload WLED_0.14.0-B1_ESP32.bin to the board using ESP Tool.  I was then able to connect to it via WiFi and program a static IP address to it.  I then updated the board with the WLED_0.14.0-b1_ESP32_Ethernet.bin file.

It still connected via WiFi, so I changed the pull down for Ethernet Type (I believe) to ESP32 POE(?) under WiFi Setup, and I lost connection to the board.  I'm guessing I chose the wrong type, but I'm not sure which type to choose.

Thanks

### To Reproduce Bug

Install "_Ethernet" version of software and change the setting noted above.

### Expected Behavior

Connection via Ethernet port.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0-b1_ESP32

### Which microcontroller/board are you seeing the problem on?

Other

### Anything else?

https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE?spm=a2g0o.detail.1000023.10.19cc1d91LNvv7g

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

cheeseshark, 2023-03-20:
Secondarily, is there a way to recover from it's current state and regain access to WLED, or do I have to reload it?

blazoncek, 2023-03-20:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

First you need to identify Ethernet chip used on your board. If it is LAN8720A the it is supported otherwise, unfortunately, no.
Second, if it is LAN8720A you will need to determine GPIO pins used by it. Verify with the manufacturer.
Once you get the GPIOs check source code which of the existing Ethernet boards uses the same GPIOs and assign that. If none are using the same GPIO then you will need to modify the source and compile yourself.

Aircoookie, 2023-03-20:
This board should be supported. Please try selecting "QuinLED-Dig-Octa" from the Ethernet type dropdown, as that board shares pin configuration with the LilyGO board.