wled/WLED #3141: Feature request: Audio input over UDP (from PC or Raspberry Pi)

WouterGritter, 2023-03-19:
**Is your feature request related to a problem? Please describe.**
I have various WLED esps/strips in my living room, and it would be very annoying to run an analog audio cable to every esp.

**Describe the solution you'd like**
A UDP audio input stream function, just like the sync function.

**Describe alternatives you've considered**
I have written a python script that uses the UDP sync function to update my strips at 50-100hz, based on an audio input on a raspberry pi (connected to my tv to sync with music/my sound bar). This works great and amazing, and is very instant, however this way I have to reimplement every effect that uses the audio input, and I cannot use these effects on WLED itself. Seeing that such a datastream gets handled perfectly (I'm using ESP8266's) I think that streaming the audio would also work great (maybe with some preprocessing on the "server" side with the audio input)

**Additional context**
I would be willing to implement this feature myself, so some pointers on where to look, what to look out for and things to consider would be awesome. I'll also update this issue with a link to my python script if other people would like to try it out!

Since WLED currently works with an analog audio input on the esp8266, I'm wondering if you could also make this work with a separate esp8266 for the audio input and UDP streaming. Currently, my setup involves a raspberry pi running a python script.

softhack007, 2023-03-19:
Hi,
Audio streaming over network is not planned yet, but let's see what the future may bring.

Actually you already have a few options already, depending on the kind of ESPs you use:
1) use [UDP sound sync](https://github.com/atuline/WLED/wiki/UDP-Sound-Sync#setup). This requires the audioreactive usermod on all devices, so its (currently) not an Option for ESP8266.
   a) from an ESP32 with audioreactive usermod and I2S microphone, to other ESP32 devices with audioreactive usermod but no physical audio input.
   b) from an external computer to ESP32s with audioreactive usermod. See our [docs about the UDP sound sync format](https://github.com/atuline/WLED/wiki/UDP-Sound-Sync#v2-format---wled-version--0140-including-moonmodules-fork). There is also a [library contributed by a user](https://github.com/netmindz/WLED-sync) that might help.

2) Use one ESP32 with audioreactive usermod, and [control others devices (ESP32 or ESP8266) via DDP](https://kno.wled.ge/advanced/ddp/). Instead of sending audio, this directly sends the LED animations to all devices.

The downside of UDP sound sync - which is based on the multicast protocol - is that it depends on the capabilities of your Wifi router. Many "consumer grade" routers cannot support multicast reliably, or will even block the protocol after a few minutes. DDP might be more reliable, depending on your router.

The audioreactive usermod (and UDP sound sync) is currently only available for ESP32, but we are working on a reduced audioreactive for 8266, which would bring UDP sound receive to these tiny old chips.

DjordjeMandic, 2024-03-01:
> The downside of UDP sound sync - which is based on the multicast protocol - is that it depends on the capabilities of your Wifi router. Many "consumer grade" routers cannot support multicast [...]

I guess its fine then with ethernet?

softhack007, 2024-03-02:
> I guess its fine then with ethernet?

I'm not sure. In case the router has problems with multicast, then connecting to the same router by ethernet will not solve the problem.

closed 2026-03-10 by softhack007