wled/WLED #3143: Make segments individually on/off when controlled with diyHUE

Aaroneisele55, 2023-03-21:
**Is your feature request related to a problem? Please describe.**
I am annoyed by the fact that you can't turn off individual segnents from external systems such as diyHUE.

**Describe the solution you'd like**
The segnents should have extra on/off state as they have in the "segments" page in the UI

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

blazoncek, 2023-03-21:
Segments have API that allows them to act independently.
But one has to have HUE (or compatible) hardware to develop WLED integration.

Aaroneisele55, 2023-03-21:
> Segments have API that allows them to act independently.
> But one has to have HUE (or compatible) hardware to develop WLED integration.

They (currently) have all features except turning on/off individual segments.
If this would be implemented, external apps could handle multiple segments much better

blazoncek, 2023-03-21:
You are welcome to add implementation if you own HUE.

Aaroneisele55, 2023-03-22:
> You are welcome to add implementation if you own HUE.
Unfortunately, I'm not that good at progrmming and imto do this
>But one has to have HUE (or compatible) hardware to develop WLED integration.

Aaroneisele55, 2023-03-22:
>But one has to have HUE (or compatible) hardware to develop WLED integration.
No, it also works if you only have WLED and a computer on which diyHUE runs (can also be a Pi Zero W on which I run it)

blazoncek, 2023-03-22:
I have no intention of using HUE or diyHUE. 😄

closed 2023-03-22 by Aaroneisele55