wled/WLED #3147: Adjustable Random Cycle Color Time

DarthWeber, 2023-03-23:
Random Cycle Color Time seems to be fixed to 5s  - which results in restless color changes in my opinion. It would be great to have this time adjustable e.g. simple in the web control like transisition times.

blazoncek, 2023-03-23:
Where would you expect to set this?

DarthWeber, 2023-03-23:
Hi, maybe under LED Settings where you set the transisiotn time ?

blazoncek, 2023-03-24:
[image]
Seems to work well. Will be available shortly.

DarthWeber, 2023-03-25:
Great ! Thanks a lot !

closed 2023-03-25 by blazoncek

DarthWeber, 2023-03-31:
Thank you very much - but is it possible to extend the setting to uint16 ? Now only times until 255s are possible. I would like to enter times for a couple of hours.

blazoncek, 2023-03-31:
You did not specify that in your original request. 🤷‍♂️ 
Fork and adjust until I get another go at it.

DarthWeber, 2023-03-31:
ah okay, sorry. I didnt expect this to be limited to a byte.