wled/WLED #3148: Home_assisant ServiceCall being made 190,000 times after inclusion in scene

chicagoandy, 2023-03-26:
### What happened?

A HomeAssistant automation is triggerinig a scene, and that scene includes WLED with 3 segments.

When that automation executes HomeAssistant crashes.   The home-assistant_v2.db Size is 30MB larger than before.

Reading through HomeAssistant events table, I see 190K 'service_call' records with the following payload: "{"domain":"light","service":"turn_on","service_data":{"transition":60.0,"brightness":3,"effect":"Twinklefox","rgb_color":[255,0,0],"entity_id":["light.ellie_bed_lights","light.ellie_lower_bed_lights","light.ellie_desk_light"]}}"

And yes, those are the three segments.    Possibly this is a HomeAssistant issue, but I doubt I'm the first to encounter it.

Anyone know why HomeAssistant is calling this service 190K times?

Firmware: 0.13.3
Hardware: esp8266

### To Reproduce Bug

A HomeAssistant automation activates a scene.   That scene contains a 3 segment WLED automation.    That service_call gets executed 190K times.

Scene Definition: 
- id: '1649851686816'
  name: Ellie - Bedtime
  entities:
    switch.ellie_heart:
      icon: mdi:heart-circle
      friendly_name: Ellie heart
      state: 'off'
    light.framelight:
      effect_list:
      - Android
      - Aurora
      - Blends
      - Blink
      - Blink Rainbow
      - Bouncing Balls
      - Bpm
      - Breathe
      - Candle
      - Candle Multi
      - Candy Cane
      - Chase
      - Chase 2
      - Chase 3
      - Chase Flash
      - Chase Flash Rnd
      - Chase Rainbow
      - Chase Random
      - Chunchun
      - Colorful
      - Colorloop
      - Colortwinkles
      - Colorwaves
      - Dancing Shadows
      - Dissolve
      - Dissolve Rnd
      - Drip
      - Dynamic
      - Dynamic Smooth
      - Fade
      - Fairy
      - Fairytwinkle
      - Fill Noise
      - Fire 2012
      - Fire Flicker
      - Fireworks
      - Fireworks 1D
      - Fireworks Starburst
      - Flow
      - Glitter
      - Gradient
[... 385 lines cut ...]
      entity_only: true

### Expected Behavior

Service_call should called exactly once.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Thread started in HomeAssistant Community Forum with screeshots:  https://community.home-assistant.io/t/help-event-firing-160-000-times/552343/1

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-03-26:
Not a WLED bug.
Pleas use Home Assistant's Github issue tracker.
Closing.

closed 2023-03-26 by blazoncek

chicagoandy, 2023-03-26:
Home Assistant https://github.com/home-assistant/core/issues/90304 logged to track the issue.