wled/WLED #3150: White Channel slider missing from >v14.

Shaps-77, 2023-03-26:
### What happened?

When using the 'PWM White' option in LED preferences, the 'White Channel' slider is not present on the UI from version 14.0 onward.
Without this slider, you are unable to drive the channel.

The following pictures show the bug.

[image]
[image]
[image]

The following picture show the correct functionality when compiling from v13.3

[image]

Thanks as always Dev team 💯 

### To Reproduce Bug

Check UI for missing 'White Channel' slider  when using 'PWM White' in LED preferences.
I think the bug is only present from version 14.0 onward..

### Expected Behavior

'White Channel' slider should be present in UI.

### Install Method

Self-Compiled

### What version of WLED?

v14 onward

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-03-27:
Please use latest source 2303240 and clear browser cache and local storage.

Shaps-77, 2023-03-27:
You're the man Blaž 🍻
All working fine with build 2303240 👍

closed 2023-03-27 by Shaps-77