wled/WLED #3152: Feature request: Flip LED output upside down (in addition to Reversed (rotated 180°))

pojlFDlxCOvZ4Kg8y1l4, 2023-03-30:
**Is your feature request related to a problem? Please describe.**
I have a 60 LED ring "analog" clock with the LEDs aimed to the wall.
This orientation is the intended use of the manufacturer.
The WLED analog clock itself works fine. NTP works fine.
However, due the orientation of the LED ring, the time is not displayed correctly. 
The LED output needs to be flipped both upside down and left to right.
LED Preferences option "Reversed (rotated 180°)" works fine for left to right.
But there's not option to flip it upside down. 
So the time is still displayed incorrectly.
I would like a toggle to flip the LED output upside down, so that the time will show correctly on the LED ring clock.

**Describe the solution you'd like**
An additional toggle like the "Reversed (rotated 180°):" under LED Preferences, but for upside down. 

**Describe alternatives you've considered**
I've tried both WLED 0.13.3 and 0.14.0-b1, both releases do not offer flipping the output upside down.
I've fiddled around with segments which was suggested in some forum posts.
This did not fix my issue.
I've searched for usermods, but could not find anything within my skillset to fix my specific usecase.
I've thought about flashing the original firmware back on it, but WLED is impressive, seemingly bug-free and feature rich whereas the original firmware is not. 
So I'd see reverting to original FW as an absolute last option. 

**Additional context**
It's a Velleman K2400 Brightdot clock.
Manual, assembly guide and the firmware zip file:  
https://www.velleman.eu/products/view?id=449304&country=us&lang=en  
Source code is available on github:
https://github.com/Velleman/K2400-BrightDot-Clock

softhack007, 2023-03-30:
In case of a LED ring, you can use "offset" to move the point where effects start. This is effectively a rotation.
Did you try that already?

[image]

softhack007, 2023-03-30:
> Effects do respect the settings, I see expected behaviour from editing segments for effects.
> Just not for the analog clock.

Then maybe its a problem with the analog clock overlay. 

@Aircoookie, @blazoncek could it be that the clock feature is "bypassing" the segments API ? Which probably means the clock overlay code should be corrected?

blazoncek, 2023-03-30:
I have no idea how to "flip" 1D LED strip "upside down". A LED ring is still just 1D strip arranged in circle.

Modifying analog overlay (and its set up) would be the only way to overcome this issue.

pojlFDlxCOvZ4Kg8y1l4, 2023-03-30:
I've swapped the power and data lines from top to bottom.  
Didn't expect much to change, but eh, worth the try I guess.   

I've made two pictures next to an actual analog clock, maybe that also helps understanding the problem.

default settings, not reversed 
[image]
reversed 180 degrees
[image]

blazoncek, 2023-03-30:
It is as simple as rotating the circle 180°.

pojlFDlxCOvZ4Kg8y1l4, 2023-03-31:
It's about the analog clock overlay. 
It is not adjustable with segments like normal effects are. 

I'd like to flip or rotate the clock adittionally to the 180 degrees reversed option under LEDE preferences, but upside down instead of left to right.  
It seems that the segment offset would do this. 
But segment offset is ignored by the analog clock overlay. Effects do listen to segment offset.  

LEDmaps look like wizardry to me, I do not understand how I would achieve my goal that way. 
Would I just create a LEDmap counting down to achieve something similar to rotating 180 degrees?
example ledmaps
`{"map":[59,58,57...2,1,0]}` ?

blazoncek, 2023-03-31:
Just rotate the circle with LEDs 180 degrees.

blazoncek, 2023-03-31:
> I've tried hot glueing some neodymium magnets to the clock. They hold for now, but it does not look like that will also pass the test of time.

Use "Green Stuff" or similar epoxy putty combined with super/instant glue.

As noted above, you cannot "rotate" 1D strip the other way as it is currently implemented. What you need is mirrored analog clock overlay.

Either talk to usermod creator(s) or wait until someone updates internal overlay.

softhack007, 2023-04-01:
> If possible I'd like to get some form of option to rotate the LEDs in software

Did you ever try what this does 🤦‍♂️ ? Changing the 12h position seems to be exactly what you ask for.
[image]

softhack007, 2023-04-01:
This is a request for a feature that already exists. 🤣 a few brains burned about nothing. But at least we had a talk.

Closing as "nothing to do".

closed 2023-04-01 by softhack007