wled/WLED #3154: “Brighter” white management set to warm kills blue

drewcovi, 2023-04-01:
### What happened?

I’m sure this behavior is as designed however I don’t think it aligns with user expectations:

Using the “Brighter” White Management setting helps bolster the white, and really get the most light out of your strip. However when white balance is set to max warmth (0% on the slider), changing the color to Blue turns the entire strip off. 

Fixing simply requires dragging the white balance to 50% or greater, however this isn’t super obvious. It seems like the White Balance value should really only come into play for White.

### To Reproduce Bug

1. Set White Management to Brighter for an rgbw strip
1. Set White Balance slider to 0% (max warmth)
1. Set color slider to maximum blue

### Expected Behavior

Strip turns blue as it has no trouble with all other colors. However it just turns off, until you drag the White Balance to the right (50%
-100%). 

### Install Method

Binary from WLED.me

### What version of WLED?

13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

This project is incredible. Sent a thank you via PayPal. I’m learning C++ and Rust so hopefully I’ll be more equipped to PR in the future. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oOJoshOo, 2023-05-06:
Same problem with accurate and the "light white" command from any Assistant.

blazoncek, 2023-05-30:
Have you considered this is intentional?
White-balance slider at max warmth is equivalent to about 2000K which does not include any blue in the spectrum.

drewcovi, 2023-05-31:
Yep, first thing I mentioned is it does seem intentional and by design.

Question is whether that meets the users expectations.

If a user is dragging a slider for color and the light simply turns off, I'd be surprised if most would even remember they had set the white temperature picker. Seems that in most use cases, color temperature applies to white exclusively, and the color wheel is the most accurate when not factoring in that setting.

closed 2024-04-23 by github-actions[bot]