wled/WLED #3159: MQTT not working correctly with brightness - colour is working fine.

877dev, 2023-04-04:
### What happened?

When in web interface and dragging brightness slider, it does not update MQTT topic with the value (topic `wled/6064f0/g`).

When publishing MQTT value 0-255 to topic i.e. `wled/6064f0` the LED's do respond but not correctly, i.e. 255 does not give full brightness, more like half brightness. Also the web interface slider is not updated.

Note the color is working fine.

### To Reproduce Bug

Drag slider in web interface and check value published to MQTT.

Publish value to topic and notice slider does not move, and LED's do not change brightness correctly.

### Expected Behavior

MQTT and web interface to be synchronised and update each other and LED correctly.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b1 (build 2212222)

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Anything else?

NodeMCU ESP-32S v1.1 and Neopixel ring.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

877dev, 2023-04-13:
@dosipod thank you, I'll look into compiling the latest firmware, I wonder if there's somewhere on the site they are posted already?

blazoncek, 2023-05-27:
My own testing showed correct behaviour, both sending and receiving.

closed 2023-05-27 by blazoncek