wled/WLED #3163: DMX effect mode changes not reflected in web UI

netmindz, 2023-04-07:
### What happened?

If you make any changes using dmx while in effect mode, these changes are not reflected in the web UI nor is there a visual indicator that DMX effect mode is active. Unlike the realtime modes that provide overlay and revert back to UI values on disconnect

### To Reproduce Bug

Make change to pattern, brightness etc via artnet or e1.31 and observe the web ui

### Expected Behavior

UI should reflect current state

### Install Method

Binary from WLED.me

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mxklb, 2023-06-08:
UI is to slow to do this as fast as DMX signals are.. see also https://github.com/Aircoookie/WLED/issues/3239

When using DMX or any other external Software to control WLED, why do you need WLED UI updates?

IMO this isn't needed, explicitly avoided because it would unnecessary slow down DMX controls.

blazoncek, 2023-06-09:
Updating interfaces has adjustable (albeit only at compile time) cooldown which prevents depleting resources and hogging CPU.

Relevant macro/constant is `INTERFACE_UPDATE_COOLDOWN`.

netmindz, 2024-04-15:
@mxklb I think the key point here is that I am talking specifically around the use of effect mode, not multi-rgb.

Therefore while yes you are still getting DXM frames many time per second, the actual changes are to any values and updates are likely to be far slower, only when the users selects say a different effect. 

This is about consistency, if you are making WLED into a "dumb" pixel pusher then we display the realtime overlay to make it clear that none of the controls can be used. When in effect mode, if I change the effect for example with Home Assistant, that is reflected in the UI, if I change the effect from DMX, then that change should also be reflected in the UI, rather than be outputting the correct effect while the UI still shows the previous

mxklb, 2024-04-15:
I think when using WLED as a DMX fixture with DMX Hard-/Software none of the DMX controls must be reflected in UI. Which is, I think the major use case for DMX. At least it's a rare requirement for people that like to control WLED pixels together with other "commercial" DMX fixtures, as I do. DMX normaly has it's own user interfaces, consoles, buttons, software .. For me it's like a remote control version of WLED, you only need WLED UI once, for setup & configuration. I never had the requirement to look into the UI, to f.e. check effects or anything else. To me it makes no sence ..

> This is about consistency, if you are making WLED into a "dumb" pixel pusher then we display the realtime overlay to make it clear that none of the controls can be used.

This is the case! So every UI user is aware that DMX remote signals are in control. I think is's fine as it is. 

P.S. Maybe UI realtime overlay depends on actively sending ArtNet or sACN and sync settings [timeout](https://kno.wled.ge/features/settings/#sync-settings).

netmindz, 2024-04-15:
While yes you shouldn't be using the UI to make changes while dmx is active, but it looks like you can if mode is effect rather than multi rgb.

When users are testing their dmx setup when in effect mode they expect to see the changes on screen that match what the actual LED output is

mxklb, 2024-04-15:
> but it looks like you can if mode is effect rather than multi rgb.

I don't use multi rgb, but when I use effect mode UI is blocked with that real time pop up you mentioned ..

If not this may be a bug

netmindz, 2024-04-15:
It's quite old, so might possibly have been fixed already. 

Related to this is that users have noticed that changes made by dmx don't replicate over using sync, but I'm less fussed by that

closed 2024-08-23 by github-actions[bot]