wled/WLED #3165: Issues with the effects page

Kentro1989, 2023-04-08:
### What happened?

When I try to scroll through the effects page the intensity bar and timer bar pull down the effects options and they all disappears

### To Reproduce Bug

I enter the app and go to the effects tab and scroll

### Expected Behavior

To scroll through the effects options without a problem

### Install Method

Binary from WLED.me

### What version of WLED?

V0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-04-09:
Use web browser. It may be a limitation of webview mobile app uses.
You may also want to try some unofficial apps if you do not want to use browser.

blazoncek, 2023-04-19:
Please use web browser (Safari) and compare it also with Firefox, Opera or any other phone browser.

roninniagara, 2023-04-21:
Yep. Same here. Seems to be a formatting issue. :(

Uofuforever, 2023-04-28:
Is there a fix out there for this without using browser?

blazoncek, 2023-04-28:
No, until app is updated.
Meanwhile you can use alternate "WLED Native" app by @Moustachauve

closed 2023-04-28 by blazoncek