wled/WLED #3166: Custom palettes with higher ids not loading unless all palettes with lower ids are present

werkstrom, 2023-04-09:
### What happened?

If a paletteX.json file is present it will not get loaded unless there are correspondning files for all lower ids. Example: If palette3.json is present, it will not load as a custom palette unless palette 0 through 2 is also present. Propose code should look for all files 0-9 and not stop once it does not finds the first missing ID.

### To Reproduce Bug

Upload (or create  a file palette3.json. Reboot. Palette won't load. Add palette files 0-2, reboot. Now custom 0-3 is present

### Expected Behavior

All present files should be loaded even though som "lower" are missing.

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.0-b2 "Hoshi" (2303240)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Uploaded palettes from the editor works as expected as long as all lower ids are present :D

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-04-09:
That is the intended behaviour as palettes are internally "pushed" into vector structure and gaps are unwanted as they waste RAM.
I do not believe it will be "fixed".

werkstrom, 2023-04-09:
Makes sense. But I would fear there will be a lot of Qs about why it does not work. It appears the custom palettes loaded also does not show up in the API, here http://[ip]/json/pal if they did (or there is another way of reading currently loaded custom palettes)  that could be used to indicate to the user in the Webinterface that changes are needed.

blazoncek, 2023-04-09:
You can get info on how many palettes there are already present and build upon that.
Number of custom palettes is available in `cpalcount` property in `info` object.

werkstrom, 2023-04-09:
So limiting the available positions in the ID selector to current max + 1 would make this easy to understand I guess. I'll look into it.

werkstrom, 2023-04-09:
Since the cpalcount value won't update until reboot, locking the tool down would not be a good idea I guess. Instead added info:
_Currently in use Custom Palette IDs are 0 to 4. Note!, IDs must be in uninterupted sequece from 0 and cannot excede 9. This number will only be updated on WLED reboot (or equivalent)._

closed 2023-04-09 by werkstrom

werkstrom, 2023-04-09:
Yep, not really that big of an issue. If there is ever a way of reloading from remote, we can implent a call then. Surely not worth the effort on it's own.

blazoncek, 2023-04-09:
I intend to implement proper custom palette loading but the spark of idea how has not yet occurred. 😉