wled/WLED #3167: Show gradient of custom palette in UI

werkstrom, 2023-04-09:
It would be nice if the gradient of the custom palettes would be visable like the gradients on the standard palettes, Would be much easier to select desired palette that way.

blazoncek, 2023-04-09:
It is, unfortunately you need to clear local storage (or at least remove `wledPalx` entry) for the palette previews to be rebuilt. It is a known issue.
You can clear local storage from UI settings page.

werkstrom, 2023-04-09:
Worked fine... Any way to do that via API?

blazoncek, 2023-04-09:
I am looking at https://www.sohamkamani.com/javascript/localstorage-with-ttl-expiry/ to implement TTL for some local storage elements to expire after a while.

werkstrom, 2023-04-09:
Aha... Would perhaps some sort of indicator that can be controlled from the "server" (WLED software) be an alternative? Like if you have a property in local storage that is "lastUpdate" so you know how old the data is, and then you can send a parameter with the page that suggests a reload and have a script compare and if the local data is older, then clear and reload. Example: If you set the "lastServerDataChange" to now() every time you reboot. And then check that time, compared to the local data you have. Could even be something in the UI like "The server suggests clearing you local storage, klick here to do it". I would imagine there are times (like when setting up) that you would like to reload this quite often and then long periods where nothing really changes... Which is why a TTL solution might not give the best experionce, IMHO

werkstrom, 2023-04-12:
Lets close, not enought upside to justify effort

closed 2023-04-12 by werkstrom

blazoncek, 2023-04-12:
Once uploading custom palette via #3164 is possible I will update palette preview logic so that new palettes are reloaded.