wled/WLED #3168: Possible to name custom palettes

werkstrom, 2023-04-09:
The ability to name custom palettes would be nice. Maybe extend the JSON to accept a "name": value. IMHO only viable if relativlely easy to implement in WLED. Added value is "nict-to-have" only. It would be easy to implement in the Custom Palette Creator.

blazoncek, 2023-04-09:
IMO Unnecessary waste of precious RAM on ESP8266.

werkstrom, 2023-04-09:
Honest Q: Implementing this in a way that if you send the name (or "n:") value, it would be added, but otherwise, current behaviour would be preserved. Would that introduce a significant increase in RAM usage? I'm thinking that since the file as of todays format would be accepted, if you have an 8266 you simply send the file in todays format. If you have an ESP32 with less RAM limits, it would peresent no major issue(?). Would the extra code in WLED to handle this be expected to be so extensive as to cause problems. To be clear: I'm not proposing this to be implemented, but asking to learn.

werkstrom, 2023-04-12:
so, lets close this then :)

closed 2023-04-12 by werkstrom