wled/WLED #3169: Live preview of custom palette while editing

werkstrom, 2023-04-09:
Add option to send current palette from Custom Palette Creator to a selected segment on the WLED device.
If calculations for each led needs to be done on the client side (browser) this would take some work on the editor. If WLED some how can accept a palette JSON as direct input via API ithat would would be much easier (from the client perspective ;) ). If this needs to be done on the client side, the approach used by Pixel Art Creator would wor, i.e. calculating each led, splitting into several API commands and sending them in sequece.

blazoncek, 2023-04-09:
That will be difficult to implement as all palettes are dynamically loaded into RAM for each segment and each frame displayed.
So effectively palette has to be uploaded to flash memory to be used for display.

werkstrom, 2023-04-09:
That's what I kind of expected. And makes sense. Second alternative then, handle it in the editor UI and sending a calculated led value for each led in the selected segment. I can see a benefit in being able to inspect the palette on the actual led strip while editing. I can get the number of LEDs via state, calculate the individual color for each led and send it directly to the API (like with Pixel Art Converter) then... I'll look in to it.

blazoncek, 2023-04-09:
There is nothing wrong by uploading it, testing uploaded version and adjust if needed. Repeat. 😄

werkstrom, 2023-04-09:
Ha ha ha... Indeed. After testing a bit here, It is really quite smooth. The device restarts in a few seconds and you're set...  Don't think the benefits will be worth the effort to be hones... Closing ;)

closed 2023-04-09 by werkstrom