wled/WLED #3172: athom music controller remote

lesleylemmens, 2023-04-12:
Since i updated my athom music controller to 0.14.0.b1 i can not londer use the remote of my controller.
I have seen somewere there is a config for this remote in a newer beta? I do not understand this, but also the sound reactive buttons will not work.
If i choose a standard 24 key remote then fuctions do not match the keys.

Does anyone have a json file for this remote? Is there a way to make all button work?

softhack007, 2023-04-12:
Hi, 
Please next time use discord for questions and support questions (links below).

The athom controller is compatible with WLED, however it has a remote and microphone that are not fully supported in standard WLED. The truth is, athom basically grabbed a copy of our source code, added their own IR, then sold the hardware+firmware package as their own product. We did not even receive a "thank you" from them.

<br/>

That said, you could be lucky because one of our devs recently "put his hand" on this controller and managed to get it going. In the [MoonModules fork](https://github.com/MoonModules/WLED), you can find a [special firmware build for athom](https://github.com/MoonModules/WLED/releases/download/v0.14.0.b15/WLEDMM_0.14.0-b15.21_athom_music_esp32_4MB_M.bin) esp32 hardware.

[image]
Join Discord to discuss the [sound reactive fork(s)](https://discord.gg/4CQRmfR) of WLED. 

[image]
Join Discord to discuss [AirCookie's WLED](https://discord.gg/KuqP7NE).

lesleylemmens, 2023-04-12:
Thank you, i will have to figure out how to use discord. 
I already found that special firmware. It works. Only the audio reactive buttons have different function.
Sorry i did not know the athom version was not officialy supported.

softhack007, 2023-04-15:
Below a picture showing how @ewoudwijma configured his IR remote in the code.
If you want another layout, its probably best to look into making [your own IR.JSON](https://kno.wled.ge/interfaces/json-ir/json_infrared/). 

[image]

IR codes are listed here:

https://github.com/MoonModules/WLED/blob/244a6137d23f0795af3c9809b28171a2c5e194b4/wled00/ir_codes.h#L215-L240

softhack007, 2023-04-15:
Closing, as its not planned to explicitly add more remotes into our source code. Custom IR.JSON can be used to configure all IR buttons as you like.

closed 2023-04-15 by softhack007

xkvmoto, 2023-10-29:
Attached is the ir.json file I use for the Athom Addressable Strip Controller  (use JSON Remote option in LED Preferences)
[ir.json](https://github.com/Aircoookie/WLED/files/13199036/ir.json)

softhack007, 2023-11-13:
> Attached is the ir.json file I use for the Athom ir Remote for the Addressable Strip Controller (use JSON Remote option in LED Preferences) [ir.json](https://github.com/Aircoookie/WLED/files/13199036/ir.json)

@xkvmoto would you mind to make a PR, to add your IR.json into our IR JSON collection? 

https://github.com/Aircoookie/WLED/tree/main/usermods/JSON_IR_remote

LutraVulgaris, 2024-03-24:
go to https://github.com/athom-tech/Sound-Reactive-WLED/releases/tag/0.13.3 and download https://github.com/athom-tech/Sound-Reactive-WLED/releases/download/0.13.3/Athom_soundReactive.bin

install it OTA (enter default password (wledota) and enable OTA updates

now you have the correct firmware again on your athom wled, with the use of the remote (enable the correct 24 key remote with music in the LED preference settings)