wled/WLED #3176: "Rainbow" Effect hangs up the Server

bemilamedia, 2023-04-16:
### What happened?

Hi all, I have a strange problem here with the "Rainbow" effect. As soon as I activate this, the web server/interface hangs up and is no longer accessible. So the interface is no longer available via the browser. This is also indicated to me with a small note in a red window "...not available". I also see this in the status in HomeAssistant. It also says "not available" after about 15 to 20 seconds at the most. However, the Rainbow effect continues.
I have also tried other effects. There is no "hang-up" of the web server/the interface. 

Even at full brightness, nothing happens in solid mode or other colour modes. At first I thought that the power supply might be the reason for the voltage collapsing for a short time. But even at a brightness of 10, this only happens in the "rainbow" effect.

I use WLED on a "Lolin Wemos D1Mini V4". There are 303 LEDs (WS2813) installed in a long string. There are several 5V power supply points available. A power supply unit with 5V20A is connected.

Even a "ping" to the IP number remains unanswered with a timeout. The ESP is simply no longer accessible, but continues to run with the rainbow effect.

### To Reproduce Bug

Switch to Rainbow Effect

### Expected Behavior

HOw to fix this? Is ist a Firmware issue or a config/setup issue?

### Install Method

Binary from WLED.me

### What version of WLED?

WLED version 0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-04-16:
I've just tried the Rainbow effect on my esp32, it works for me without side-effects on the web interface. 

* How did you set effect  speed and intensity sliders?
* which GPIO pin are you using for LEDs? Some pins on 8266 are not having hardware acceleration and may be slow.
* Can you try without HA integration? HomeAssistent tends to poll WLED too often, which causes crash or makes the web UI unresponsive.

bemilamedia, 2023-04-16:
HI,
I have tested it with different speeds. With Effect Speed 128 and Effect Size 128 it hangs after a few seconds. If I change Effect Speed to 10, it runs for several minutes. But at some point it hangs.

I have connected the data line to GPIO 4 (pin 4).
I have now completely deactivated HA integration. It still happens. Sometimes earlier, sometimes later.

The loading time of the interface via the browser also seems a little slower when I run the Rainbox Effect. At Effect Speed 128 even a little slower than at 10.

I have now tried a few effects back and forth. And now I've been running the Rainbox Effect for 5 minutes at Effect Speed 128. Every time I refresh the web interface it loads slower (sometimes 30 seconds until the page is rebuilt or even until the timeout), so it looks like the ESP memory is filling up.

If I call up a ping on the IP of the ESP in the terminal, this is then at +-1000ms-.

blazoncek, 2023-04-16:
There's nothing wrong with Rainbow effect. 26 of my devices run it smoothly (ESP01, ESP8266 D1 mini, ESP32, ESP32-C3 and ESP32-S2, with or without Ethernet.

If you are using ESP8266 please use one of the recommended GPIO pins (1 or 2) and not GPIO4.

bemilamedia, 2023-04-17:
Changing to GPIO 2 solves the "problem.
Thanks.

closed 2023-04-17 by bemilamedia