wled/WLED #3180: Translations and timezones

LeoZanella, 2023-04-20:
Can we have a separate file for translations?
I think this would make it better to add languages.
I'm trying to translate it to Portuguese and it is being really hard.
About the timezones, wouldn't it be better if, instead of zones name, we have the increments (-3 for Brazil).
I have to configure it and a not good way to get the time right for my zone.

softhack007, 2023-06-10:
Hi, I think that the timezone name plus increment is a good idea, and will be simple to implement; for example we could show "[GMT+1] CET/CEST" to indicate the offset from GMT. Just that we would not include all the capitals in that time zone, to avoid inflating firmware size unnecessarily.

https://github.com/Aircoookie/WLED/blob/999bec19f1bf37b344177a9aeffc752c5f4ae472/wled00/data/settings_time.htm#L161-L186

About translation support .... well that's more complicated. WLED is missing the source code infrastructure to support translations, and some texts are directly coded in .cpp files and not found in the web sources inside data/.

You might be able to do a translation for yourself right now, but we cannot include that into the main WLED repo Afterwards. See also:
* https://github.com/Aircoookie/WLED/issues/459
* https://github.com/Aircoookie/WLED/issues/2165

Also as WLED is developing further, we'd need translation contributors who would take care of translating new things  with each release.

softhack007, 2023-10-07:
I'm putting the "[GMT+1] CET/CEST" idea on the list for 0.15.1, as it seems easy to implement.

For translations - sorry no news, we still have no concept/design to support that request.

Sojourneer, 2024-09-27:
I am working on this.  The flow is as follows, populating an I18N folder structure, and finally copying localized files back to wled00/data.  Each step is supported by a script:
1) Scope: Get list of wled00/data files from cdata.js
2) I18N: For each file, preserve into an I18N folder, and then scan each file for text and for @title attributes, outputting I18N templated files and a phrase index 
4) Translate: Localize the index using google translate API
5) L12N: Apply the index to the templates, and copy the localized files into the wled00/data
After this, the UI node build would be used.

I'm working on step (5) now.

blazoncek, 2024-09-27:
@Sojourneer please work on 0_15 branch.

Sojourneer, 2024-10-10:
> @Sojourneer please work on 0_15 branch.
Understood.  My target is to finish for this Christmas season. November at latest, and I'll plan to move it over by then.

Sojourneer, 2024-10-13:
I've finished the first round, up to L12N and building the UI (I'm on the road so not able to test on an actual device).
Looking at the resulting HTML, I've noticed a fair bit of dynamic generation of HTML.  I'm thinking of changing the dynamic generation scripts to pull from hidden HTML rather than strings, which would allow use of my translation apparatus.
@blazoncek Any comments on the above?

blazoncek, 2024-10-13:
@Sojourneer there are three places that provide text for UI: HTML & JS files and C++ source (in JSON response to requests). The latter are few.

As we are working on MCU that has limited resources (RAM, flash, CPU power) it would be best to limit number of I18N files to only the one required by user.

My suggestion would be to add JS code, that will load (i.e. from Github repository or from embedded flash) the localisation file and keep it in browser's local storage or cache. AFAIK getting files from Github may be problematic since there is no HTTPS support in WLED so my recommendation would be to allow user to upload localisation file to flash.
I would keep English texts in sources and replace them when UI loads.

Sojourneer, 2024-10-13:
> @Sojourneer there are three places that provide text for UI: HTML & JS files and C++ source (in JSON response to requests). The latter are few.
>
> As we are working on MCU that has limited resources [...]

I think you may have misunderstood my approach.  I am providing for localization that is applied at build-time, not dynamically.  I don't require dynamic language switching, and therefore I am not pursuing that.  The resource requirements should be minimum.

blazoncek, 2024-10-13:
IMO that is unacceptable as users will most likely download official builds *that will not be localised*.

Sojourneer, 2024-10-13:
> IMO that is unacceptable as users will most likely download official builds _that will not be localised_.

Ah well, that is too bad.  Maybe your translation community can be sweet-talked into building the necessary versions.
In the meantime, I will work only on my current fork.

Aircoookie, 2024-10-13:
Some years ago by now, I also put some thoughts into how to best support translations. I've arrived at the following conclusions back then as to what would likely make most sense.
- purely in-browser JS - this won't require any extra processing on the MCU to support I18N.
- downloading a localisation dict .json from github for the installed WLED version and desired language (HTTPS is not an issue, since the browser fetches from github, not the ESP itself). The only drawback is that internet connection is required at the time of language selection. Alternatively, the language file could be uploaded to the ESP filesystem
- store this translation file in browser localStorage

There are some details left to work out like how/when to update language files on firmware update and how to translate English strings assembled on the ESP, but that is the overall idea.

Language specific binaries are a valid approach, but I don't think they would be the best fit for the project long term as they are not flexible and would balloon the plethora of different builds that are already out there even further. It would essentially only be possible to supply all possible binary variants with a build-on-demand website that lets you pick the desired features and language, which I don't think is the best user experience and also harder to maintain. Don't get me wrong, I really appreciate you looking into translations though :)

Sojourneer, 2024-10-13:
> Some years ago by now, I also put some thoughts into how to best support translations. I've arrived at the following conclusions back then as to what would likely make most sense.
> - purely in-browser JS - this won't require any extra processing on the MCU to support I18N.
> - downloading a localisation dict .json from github for the installed WLED version and desired language (HTTPS is not an issue, since the browser fetches from github, not the ESP itself). The only drawback is that internet connection is required at the time of language selection. Alternatively, the language file could be uploaded to the ESP filesystem
> - store this translation file in browser localStorage and/or ESP filesystem
> 
> There are some details left to work out like how/when to update language files on firmware update and how to translate English strings assembled on the ESP, but that is the overall idea.
> 
> Language specific binaries are a valid approach, but I don't think they would be the best fit for the project long term as they are not flexible and would balloon the plethora of different builds that are already out there even further. It would essentially only be possible to supply all possible binary variants with a build-on-demand website that lets you pick the desired features and language, which I don't think is the best user experience and also harder to maintain. Don't get me wrong, I really appreciate you looking into translations though :)

I need the Japanese translation now, so I have no choice.  I'll update this post as I address the remaining fiddly bits.

Sojourneer, 2024-10-17:
> * purely in-browser JS - this won't require any extra processing on the MCU to support I18N.
>
> * downloading a localisation dict .json from github for the installed WLED version and desired language [...]

After trying out on a device, and taking a closer look, I think this shouldn't be too hard applying the L12N on the browser side after doing the I18N and translation preparation along the lines I have taken so far (templatizing) + listing out the effects and palettes using the provided json URL.  This would allow for dynamic translation.

Sojourneer, 2024-10-24:
@blazoncek @Aircoookie 
> > @Sojourneer please work on 0_15 branch.
> > Understood.  My target is to finish for this Christmas season. November at latest, and I'll plan to move it over by then.  Busy busy.

FYI: I am underway with the dynamic language feature, and have gotten the translation tools and templating/localization of the HTML to work.  I have made a simple python HTTPServer with .on() handlers to simulate the application enough to test the UI.
I notice a problem in GetURL of 0_15 index.js which is interfering with testing, and makes me think that branch is not yet stable + and also quite substantial.
So, with apologies, I am going to move the translation work to a fresh branch of main for the meantime.

blazoncek, 2024-10-24:
`main` is obsolete and will be replaced by `0_15` shortly.
`getURL()` is there to allow running WLED behind reverse proxy.

Sojourneer, 2024-10-24:
> `main` is obsolete and will be replaced by `0_15` shortly.
> `getURL()` is there to allow running WLED behind reverse proxy.

getURL() accesses /index.html/json/palettes etc, and I have tried with various  host/port combinations.  I was not able to reverse engineer a spec for it.  How well has 0_15 been tested?